micropython/micropython #5000: BUG: finally/continue

pmp-p, 2019-08-12:
from  https://bugs.python.org/issue37830

### cpython 3.8 will segfault on
```
def main():
    def simple():
        for number in range(2):
            try:
                return number
            finally:
                continue
    print(simple())
print("begin")
main()
print("end")
```

### micropython master will do
```
begin
Traceback (most recent call last):
  File "cont.py", line 10, in <module>
  File "cont.py", line 8, in main
  File "cont.py", line 5, in simple
NameError: local variable referenced before assignment
```

### but rustpython wins ( even javascript version ) !
```
begin
0
end
```

jimmo, 2019-08-12:
MicroPython should match 3.4, so raising the `SynaxError` would be the expected behavior.

FWIW, I don't think rustpython is correct here either? It's hard to tell from the linked bugs, but my interpretation is that `continue` inside `finally` (when implemented) should behave like `return` and `break` -- i.e. they effectively cancel the current `return`.

For example, with `break`: the following prints `22` on Python 3.7 and MicroPython, but `0` on rustpython.
```
def main():
    def simple():
        for number in range(2):
            try:
                return number
            finally:
                break
        return 22
    print(simple())
print("begin")
main()
print("end")
```

dpgeorge, 2019-08-13:
Thanks for the report.  It's hard to get this correct, we previously had bugs for both return and break within a finally, which were fixed (after a lot of effort to find a minimal way to fix them) in b735208403a54774f9fd3d966f7c1a194c41870f and e1fb03f3e2d1e99d0f745e9e17a963a036d34476

Using continue (and even return/break) in a finally is really bad practice (IMO) so, unless there's an elegant way to implement it in the VM, it should simply be banned by the compiler.

dpgeorge, 2019-09-27:
I made a PEP601 to forbid this behaviour in CPython, but it was rejected: https://www.python.org/dev/peps/pep-0601/

So for MicroPython, either the use of continue in finally raises a SyntaxError, or is fixed.

dpgeorge, 2019-09-27:
It looks like MicroPython still doesn't handle break correctly in a finally, because the following will crash it:
```python
while 1:
    while 1:
        try:
            1 / 1
        finally:
            break
```
The break and continue handling seem to have the same problem, that the stack grows.  Replacing `1/1` with `1/0` shows the error in a different way.

dpgeorge, 2019-09-27:
See #5154 for a proposed fix.