micropython/micropython #5005: Potential assertion failure in mp_obj_tuple_get with tuple "subclasses"

jepler, 2019-08-13:
In Circuitpython, we encountered a crash when using a namedtuple to represent a time.struct_time, because other layers of the code used `mp_obj_tuple_get` which requires exactly a tuple:
```
Breakpoint 1, __assert_func (file=0x8db34 "../../py/objtuple.c", line=254, 
    func=0x9a9b0 <__func__.12400> "mp_obj_tuple_get", 
    expr=0x8db0c "MP_OBJ_IS_TYPE(self_in, &mp_type_tuple)") at ../../main.c:486
486	    mp_printf(&mp_plat_print, "Assertion '%s' failed, at file %s:%d\n", expr, file, line);
(gdb) where
#0  __assert_func (file=0x8db34 "../../py/objtuple.c", line=254, 
    func=0x9a9b0 <__func__.12400> "mp_obj_tuple_get", 
    expr=0x8db0c "MP_OBJ_IS_TYPE(self_in, &mp_type_tuple)") at ../../main.c:486
#1  0x00047dea in mp_obj_tuple_get (self_in=0x200081a0, len=0x2003fd38, 
    items=0x2003fd3c) at ../../py/objtuple.c:254
#2  0x0006e7a0 in struct_time_to_tm (t=0x200081a0, tm=0x2003fd48)
    at ../../shared-bindings/time/__init__.c:173
```
In CIrcuitPython we addressed it by changing the condition in `mp_obj_tuple_get` to match `tuple_cmp_helper`:
```
    // type check is done on getiter method to allow tuple, namedtuple, attrtuple   
    mp_check_self(mp_obj_get_type(self_in)->getiter == mp_obj_tuple_getiter);
```

@dpgeorge suggests that instead the example in `objdict.c` should be followed:
```
#define mp_obj_is_dict_type(o) (mp_obj_is_obj(o) && ((mp_obj_base_t*)MP_OBJ_TO_PTR(o))->type->make_new == dict_make_new)
```
however, rather than comparing the specific type `make_new` method (which is `namedtuple_make_new` for a namedtuple), I assume to still compare `getiter` methods in a new macro called something like `mp_obj_is_tuple_compatible`.

stinos, 2019-08-13:
> using a namedtuple to represent a time.struct_time

Can you show some code explaining what this means exactly?

jepler, 2019-08-13:
Whoops, I meant to include this reference to the issue in circuitpython: https://github.com/adafruit/circuitpython/issues/2052

Here's how we coded `time_to_tm`:
```
void struct_time_to_tm(mp_obj_t t, timeutils_struct_time_t *tm) {
    mp_obj_t *elems;
    size_t len;

    if (!MP_OBJ_IS_TYPE(t, &mp_type_tuple) && !MP_OBJ_IS_TYPE(t, MP_OBJ_FROM_PTR(&struct_time_type_obj))) {
        mp_raise_TypeError(translate("Tuple or struct_time argument required"));
    }

    mp_obj_tuple_get(t, &len, &elems);
    if (len != 9) {
        mp_raise_TypeError(translate("function takes exactly 9 arguments"));
    }

    tm->tm_year = mp_obj_get_int(elems[0]);
    tm->tm_mon = mp_obj_get_int(elems[1]);
    tm->tm_mday = mp_obj_get_int(elems[2]);
    tm->tm_hour = mp_obj_get_int(elems[3]);
    tm->tm_min = mp_obj_get_int(elems[4]);
    tm->tm_sec = mp_obj_get_int(elems[5]);
    tm->tm_wday = mp_obj_get_int(elems[6]);
    tm->tm_yday = mp_obj_get_int(elems[7]);
    // elems[8] tm_isdst is not supported
}
```
we wanted to uniformly use ordinary tuples, or namedtuples.  Having `time.struct_time` increases compatibility with cpython3 for us.

jepler, 2019-08-13:
It does look like none of hte uses of mp_obj_tuple_get in micropython would have this problem, they are either deeply internal to the type system (maybe you *COULD* make one of these fire with metaclass nonsense if you tried) or already guarded with a check for tuple type, like in `objstr.c`:
```
        if (mp_obj_is_type(rhs_in, &mp_type_tuple)) {
            // TODO: Support tuple subclasses?
            mp_obj_tuple_get(rhs_in, &n_args, &args);
```

jepler, 2019-08-13:
I couldn't find any paths to make this hit the assertion error in circuitpython.  I did discover some python3 things that do not work in micropython, but they are of little importance.

python3:
```
>>> t = collections.namedtuple('t', ('b',))
>>> type('x', t(object), {})
<class '__main__.x'>
```

micropython:
```
>>> t = ucollections.namedtuple('t', ('b',))
>>> type('x', t(object), {})
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 
```
```

commit 12f13ee6 (2019-08-30): py/objtuple: Allow compatible subclasses of tuple in mp_obj_tuple_get.

As part of this patch a private macro mp_obj_is_tuple_compatible is
introduced to encapsulate the check, which is used in two locations.

Fixes #5005.
closed 2019-08-30 by dpgeorge