micropython/micropython #5010: Pyboard D: space in flash for frozen bytecode

peterhinch, 2019-08-15:
This is limited compared to the Pyboard 1.x. In order to compile for the SF2W I had to minimise the number of files in the modules directory otherwise (after ensuring a clean build) I get
```
... successful output omitted ...
LINK build-PYBD_SF2/firmware.elf
arm-none-eabi-ld: build-PYBD_SF2/firmware.elf section `.text' will not fit in region `FLASH_APP'
arm-none-eabi-ld: region `FLASH_APP' overflowed by 15724 bytes
Makefile:538: recipe for target 'build-PYBD_SF2/firmware.elf' failed
make: *** [build-PYBD_SF2/firmware.elf] Error 1
Build failure
```
Is this a hardware limitation or is there a way to specify more space for frozen bytecode?

dpgeorge, 2019-08-15:
> Is this a hardware limitation or is there a way to specify more space for frozen bytecode?

There is space (up to 2mbyte) in the external flash which is memory mapped.  To put frozen code in this location requires the following patch:
```diff
--- a/ports/stm32/boards/PYBD_SF2/f722_qspi.ld
+++ b/ports/stm32/boards/PYBD_SF2/f722_qspi.ld
@@ -48,6 +48,7 @@ SECTIONS
     .text_ext :
     {
         . = ALIGN(4);
+        *frozen_mpy.o(.text* .rodata*)
         *lib/mbedtls/*(.text* .rodata*)
         . = ALIGN(512);
         *(.big_const*)
```

peterhinch, 2019-08-15:
That works fine. Thank you.

peterhinch, 2019-08-18:
There does seem to be a performance penalty incurred by doing this, presumably because the MicroPython VM runs from external rather than internal flash. In my test Python code took 33% longer to execute. This is regardless of whether the Python code is frozen or located in the filesystem.

jimmo, 2019-08-18:
I would expect the three cases to have differing performance:
 1 - frozen, bytecode in internal flash
 2 - frozen, bytecode in external flash
 3 - all other cases (.py or .mpy on filesystem), bytecode in ram

Although I'd hope that L1 caching on the M7 would somewhat make the difference fairly small!

But the patch in https://github.com/micropython/micropython/issues/5010#issuecomment-521636720 should move _only_ the code for the frozen modules into external qspiflash (i.e. case 1 -> 2 above). Everything else in the `.text` segment (including the vm itself) should be unchanged, so it's interesting that performance of filesystem code was affected.

You can verify what is in external flash by looking at the elf with `nm build-PYBD_SF2/firmware.elf` -- symbols at 0x08... are internal, 0x9... are external.

robert-hh, 2019-08-18:
The code size for the SF2 module exceeds the internal flash. So there must be code in external flash, which is subject to delays, caused by swapping.

dpgeorge, 2019-08-19:
> There does seem to be a performance penalty incurred by doing this

Yes you're right, I forgot about this.

The reason is that part of the qstr table now lives in external flash, the frozen qstrs.  And this leads to external SPI flash accesses for any dereference of any qstr (eg to get its hash to look it up in a dict, or get its string value) because the `qstr_pool_t` that lives in SPI flash must be accessed.

It should be possible to place the frozen qstrs in flash (at least just the index table, not the actual data) to get back the performance.

robert-hh, 2019-10-01:
Today's build with the improved BL part caused that error on my package too. So maybe it's worth to have a look at qstr table optimization.

dpgeorge, 2019-10-05:
> Today's build with the improved BL part caused that error on my package too.

In 4a6974bea598a9b76c4a49afcc0d6f82760a7006 I moved the BLE stack library to external flash.  It's not performance critical so can easily live there.  That will get back all the internal flash that was there before for frozen bytecode.

> So maybe it's worth to have a look at qstr table optimization.

I had a look and the relevant patch is:
```diff
--- a/ports/stm32/boards/PYBD_SF2/f722_qspi.ld
+++ b/ports/stm32/boards/PYBD_SF2/f722_qspi.ld
@@ -48,6 +48,7 @@ SECTIONS
     .text_ext :
     {
         . = ALIGN(4);
+        *frozen_mpy.o(.rodata.const_obj_* .rodata.const_table_data_* .rodata.fun_data_* .rodata.raw_code_*)
         *lib/mbedtls/*(.text* .rodata*)
         *lib/mynewt-nimble/*(.text* .rodata*)
         . = ALIGN(512);
```
Running pystone, I see no performance difference with this change, putting frozen code in external flash (but not the frozen qstrs).

I don't want to put that change directly in master, I'd want to somehow have it configurable.