micropython/micropython #5013: Unable to build SF2/SF6 after pybd merged back into main repo

pacmac, 2019-08-16:
Just tried building latest with the PYBD merged back into the main repo, instead of from Damien's repo, and both SF2 and SF6 fail with similar errors:

```
CC build-PYBD_SF2/frozen_mpy.c
LINK build-PYBD_SF2/firmware.elf
arm-none-eabi-ld: cannot find build-PYBD_SF2/lib/stm32lib/CMSIS/STM32F7xx/Source/Templates/gcc/startup_stm32f722xx.o: No such file or directory
arm-none-eabi-ld: cannot find build-PYBD_SF2/lib/stm32lib/CMSIS/STM32F7xx/Source/Templates/system_stm32f7xx.o: No such file or directory
make: *** [Makefile:539: build-PYBD_SF2/firmware.elf] Error 1
```

```
CC mbedtls/mbedtls_port.c
CC build-PYBD_SF6/pins_PYBD_SF6.c
CC build-PYBD_SF6/frozen_mpy.c
LINK build-PYBD_SF6/firmware.elf
arm-none-eabi-ld: cannot find build-PYBD_SF6/lib/stm32lib/CMSIS/STM32F7xx/Source/Templates/gcc/startup_stm32f767xx.o: No such file or directory
arm-none-eabi-ld: cannot find build-PYBD_SF6/lib/stm32lib/CMSIS/STM32F7xx/Source/Templates/system_stm32f7xx.o: No such file or directory
make: *** [Makefile:539: build-PYBD_SF6/firmware.elf] Error 1
```

I have done a clean build, what am I missing ???

pacmac, 2019-08-17:
I just changed my settings back to use https://github.com/dpgeorge/micropython and there's no problem with building.

dpgeorge, 2019-08-17:
> I have done a clean build, what am I missing

You probably need to do `git submodule update` to get the latest stm32lib.

pacmac, 2019-08-17:
Thanks, all my builds, updates etc are done by custom bash scripts, and when I ran the submodule update command manually it failed to update 3 sub-modules including stmlib.

So I did a fresh re-clone and stm32-sf2, stm32-sf6, stm-pyb, nrf, esp8266 & esp32 all compile with no problems.

Closing this now.

closed 2019-08-17 by pacmac