micropython/micropython #5015: Yet another strange bug

jdcrunchman, 2019-08-16:
[quickies.py.zip](https://github.com/micropython/micropython/files/3510722/quickies.py.zip)

See the attached file I call quickies.py

>>> import quickies
quickies loaded
>>> pwda() .        <--- I call this function here to print working directory
Traceback (most recent call last):      <---- and I get this error saying 'pwda' is not defined/
  File "<stdin>", line 1, in <module>
NameError: name 'pwda' isn't defined

>>> dir(quickies)
['__class__', '__name__', 'remove', '__file__', 'cd', 'chdir', 'getcwd', 'listdir', 'rmdir', 'ls', 'rm', 'dispfile', 'removedir', 'pwda']

And as you see above,  'pwda' is definitely defined.

WTF?

Is this a bug,  and if not,  can I please ask you what I'm doing wrong?

John

jdcrunchman, 2019-08-16:
By the way,  instead of just defining a "pwda" function to print the working directory,\
I do it by hand,  in the python shell,   it works.

>>> import os
>>> os.getcwd()
'/'

So,  why can't I define and use "getcwd" in a function.   Is this some kind of restriction that is undocumented?

"A very confused microPython user"
John

robert-hh, 2019-08-16:
How  about calling quickies.pwda()

robert-hh, 2019-08-16:
May I also point your attention to:
https://github.com/pfalcon/pycopy-lib/tree/master/upysh
That module implements what you attempt to do with your file.

jdcrunchman, 2019-08-16:
I found another problem to call to your attention,  although this might be an Atom/PyMakr
issue.
I put this code in a file I call "pytest.py"

from www.webinput import *
msgadd('msgfrom', 'john', 'jason', 'hi,  this is john')
msgadd('msgto', 'jason', 'john', 'hi,  this is jason, how are things')
msgadd('msgfrom', 'john', 'jason', 'i am busy')
msgadd('msgfrom', 'Tolga', 'moga', 'Hi,  i am butting in here')
print(dispAllMessages('john'))

Earlier,  I would run it,  and get the expected results.  Using the PyMakr "run" button.

Then,  I made some changes,   which broke something,  so I commented out the bad changes
and then ran it again...

When I "ran" it,  I get...

>>> Running selected lines

>>>
>>>
>
MicroPython v1.11-180-g8f55a8fab on 2019-07-27; ESP32 module with ESP32
Type "help()" for more information.
>>>

But,  when I imported it...

>>> import webtest
<div class="msgfrom">hi,  this is john</div>
<div class="msgto">hi,  this is jason, how are things</div>
<div class="msgfrom">i am busy</div>

It works OK...   WHY when I import instead of run,  it works,  but when I 'run" it doesn't
So whats the difference between the import and the run?

John

jdcrunchman, 2019-08-16:
WOW - that output above was supposed to generate HTML code,  but it was stripped out... when I copied it into this forum....   How do I avoid this doing that..?

if I want to paste in HTML into this forum,  how do I do that without it stripping out the HTML?

FUCKin wierd...

jdcrunchman, 2019-08-16:
>>> import webtest
<div class="msgfrom">hi,  this is john</div>
<div class="msgto">hi,  this is jason, how are things</div>
<div class="msgfrom">i am busy</div>

jdcrunchman, 2019-08-16:
Ok,  FUCK this shit,   see the attached ZIP file....  to see what I REALLY want to put up here.

[to_forum.zip](https://github.com/micropython/micropython/files/3511269/to_forum.zip)

Guys,  I'm really going to need to supply HTML code in many instances of posting these bug reports,  can someone please tell me how to do it,  without this forum stripping it out

So unzip the "to_forum" file to see what I really wanted to put up here...  I don't need this fuckin bullshit.

JOhn

dhylands, 2019-08-16:
This isn't a forum. You is a bug tracker.

You should be posting in the forum over here: forum.micropython.org

Dave Hylands

MrSurly, 2019-08-16:
> if I want to paste in HTML into this forum, how do I do that without it stripping out the HTML?

Backticks.  See the editing bar.

dpgeorge, 2019-08-17:
Please use the forum for discussions like this.

closed 2019-08-17 by dpgeorge

jdcrunchman, 2019-08-17:
Ok,  got it, but some of my issues could also have come from PyMakr Atom plugin.
I think I have a repeatable bug.    Earlier,  I stated that in some cases,  I can "run" a sctript,  but then by adding just another Method definition to the file "webinput.py" which I already sent you,  and it runs just fine,  until I added a new method...   See below.

    def getMyName(self):
        return(self.me)

I added this to my "response" class,  and after adding this,  the run no longer works.
When running it,   I get...

MicroPython v1.11-180-g8f55a8fab on 2019-07-27; ESP32 module with ESP32
Type "help()" for more information.
>>>
>>> Running selected lines

>>>
>

But when importing it,   I get....

MicroPython v1.11-180-g8f55a8fab on 2019-07-27; ESP32 module with ESP32
Type "help()" for more information.
>>> Running selected lines

>>>
>>> 

I would attach the HTML I got,  but this bug tracker won't allow me to display raw HTML

To get that HTML output,  I added these lines at the end of webinput.py

from www.webinput import *
msgadd('msgfrom', 'john', 'jason', 'hi,  this is john')
msgadd('msgto', 'jason', 'john', 'hi,  this is jason, how are things')
msgadd('msgfrom', 'john', 'jason', 'i am busy')
msgadd('msgfrom', 'Tolga', 'moga', 'Hi,  i am butting in here')
print(dispAllMessages('john'))

Can you please tell me how I can send a fragment of "div" lines to this tracker software.
I tried everything under the book,  but if you still have that file I sent you titled "to_forum.zip"
it has the "div" statements it was supposed to print.

So I hope you can repeat the bug.

John

robert-hh, 2019-08-18:
Pymakr is a product of Pycom, meant to work with the product line and firmware. Some of the functionality it uses is not implemented in the micropython.org firmware. So better ask at the Pycom forum forum.pycom.io or, if you think it is a bug,  at the pymakr repository https://github.com/pycom/pymakr-atom

Robert

jdcrunchman, 2019-08-18:
i know that,  but with so many actors and integrated systems,   PyMakr, Atom, Micropython, PyCom,  any one of them could have this problem,   I just pointed it out.

MrSurly, 2019-08-18:
> Can you please tell me how I can send a fragment of "div" lines to this tracker software.

Use either single backticks for inline: \`\<div\>\`, result: `<div>`

For multiple lines, you can use triple backticks:
\`\`\`
\<div\>
 \<pre\>
  Hi there
  \</pre\>
\</div\>
\`\`\`

Result:
```
<div>
 <pre>
  Hi there
  </pre>
 </div>
```

Bonus answer, for the above examples, I used backslash to escape individual characters: \\\`\\\<div\\\>\\\`

MrSurly, 2019-08-18:
@jdcrunchman You couldn't possibly be _the_ [jdcrunchman](https://en.wikipedia.org/wiki/John_Draper)?

jdcrunchman, 2019-08-19:
I am the Crunchman,  why do you ask?    To reach me privately,  email to jdcrunchman at protonmail dot com.

jdcrunchman, 2019-08-19:
I have more strangeness using MicroPython
[quickies.py.zip](https://github.com/micropython/micropython/files/3517426/quickies.py.zip)

See my following code,  it's just used to help me look at files.

>>> import quickies
quickies loaded .     <---- as expected.

>>> dir(quickies)
['__class__', '__name__', 'remove', '__file__', 'cd', 'chdir', 'getcwd', 'listdir', 'rmdir', 'ls', 'rm', 'dispfile', 'removedir', 'pwda']

To show that these definitions have been imported in.
As you can see,  "ls" shows up in the list.

But when I run the "ls" function as below...

>>> ls()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ls' isn't defined

It says "ls" isn't defined.   
Please realize,  I'm very new to Python3,  and haven't done anything with it in more then 2 decades
so please forgive me if I make a stupid mistake,   but the way I defined "ls" was always the way I did it.

But I don't think I'm doing anything wrong,  do you?     If you don't think so,  then I have a legit bug to report.

Now,  last week,  and many weeks before,  I've been able to use this "quickies" module.
It was only today,  when now it suddenly didn't work

Thanx

dpgeorge, 2019-08-20:
> But when I run the "ls" function as below...
> It says "ls" isn't defined.

As @robert-hh mentioned above you need to either do `from quickies import *`, or use `quickies.ls()`.  This is how normal Python works with regards to namespaces and global variables.

jdcrunchman, 2019-08-20:
I never could find anything in the Docs about how to embed HTML into a response.
I had no clue about using "..." to encapsulate HTML,   thanx a lot.