micropython/micropython #5018: extmod: duplicate symbols when shared linking

pmp-p, 2019-08-18:
wasm-ld of clang 10 does *not* like STATIC defined as static for void, or most likely functions pointer address will get corrupted.

but redefining STATIC to nothing leads to some collisions:

```
wasm-ld: error: duplicate symbol: heap_siftdown
>>> defined in build/extmod/moduheapq.o
>>> defined in build/extmod/modutimeq.o

wasm-ld: error: duplicate symbol: heap_siftup
>>> defined in build/extmod/moduheapq.o
>>> defined in build/extmod/modutimeq.o
```

(heap_siftdown and heap_siftup seem identical ), maybe get_heap need special treatment.

dpgeorge, 2019-08-19:
Thanks for the report.  There are indeed a few of these cases.

Since it was intended to be able to compile with `STATIC` defined to nothing these cases should all be fixed.

dpgeorge, 2019-08-20:
Fixed by 0cc8910bc58c1a1f69a696f5efe4ccf0ffcf2984

closed 2019-08-20 by dpgeorge