micropython/micropython #5022: network.LAN on NUCLEO_F429ZI

MichaelAkhnoukh, 2019-08-19:
i'm having the same issue presented in this thread
https://forum.micropython.org/viewtopic.php?f=12&t=6804

the REPL crashes whenever network.LAN() instance is created

dpgeorge, 2019-08-20:
Unfortunately I don't have one of these boards to test it.  All development of the ETH driver was done an an F7xx MCU, although there were reports of it working on the F4.

Do you have an F7 Nucleo or Discovery board that you can test the same code on?

MichaelAkhnoukh, 2019-08-20:
> Do you have an F7 Nucleo or Discovery board that you can test the same code on?

not yet

MichaelAkhnoukh, 2019-08-20:
i've managed to fix the issue and created a pull request.

dpgeorge, 2019-08-28:
Fixed by b1e04848ef62d20db42b0b30f24304307c4e6765

closed 2019-08-28 by dpgeorge

TA1DB, 2019-09-11:
I confirm as of today network.LAN() works as expected on STM32F7DISC as well.