micropython/micropython #5023: stack smashing detected

AndreAhmed, 2019-08-19:
I'm tyring to write micropython code in STM32F779I. I get while running the app without running any code stack smashing detected.

Here is my memory layout

```
/* Highest address of the user mode stack */
_estack = 0x20080000;    /* end of RAM */

/* Generate a link error if heap and stack don't fit into RAM */
_Min_Heap_Size = 0x1000;      /* required amount of heap  */
_Min_Stack_Size = 0x2000; /* required amount of stack */

/* Specify the memory areas */
MEMORY
{
  FLASH (rx)      : ORIGIN = 0x08000000, LENGTH = 2048K
  DTCMRAM (xrw)   : ORIGIN = 0x20000000, LENGTH = 128K
  SRAM1 (xrw)     : ORIGIN = 0x20020000, LENGTH = 368K
  SRAM2 (xrw)     : ORIGIN = 0x2007C000, LENGTH = 16K
  MEMORY_B1 (rx)  : ORIGIN = 0x60000000, LENGTH = 0K
}

main_stack_base = _estack;
/* used by the startup code to populate variables used by the C code */
data_lma = LOADADDR(.data);
data_vma = ADDR(.data);
data_size = SIZEOF(.data);

/* used by the startup code to wipe memory */
ccmram_start = ORIGIN(SRAM1);
ccmram_end = ORIGIN(SRAM1) + 4;

/* used by the startup code to wipe memory */
sram_start = ORIGIN(SRAM1);
sram_end = ORIGIN(SRAM1) + LENGTH(SRAM1);
_ram_start = sram_start;
_ram_end = sram_end;

_codelen = LENGTH(FLASH);
_flash_start = ORIGIN(FLASH);
_flash_end = ORIGIN(FLASH) + LENGTH(FLASH);

_estack = ORIGIN(SRAM1) + LENGTH(SRAM1) - 8;
_sstack = _estack; /* tunable */

_heap_start = _ebss;
_heap_end = _sstack;

/* Define output sections */
SECTIONS
{
  /* The startup code goes first into FLASH */
  .isr_vector :
  {
    . = ALIGN(4);
    KEEP(*(.isr_vector)) /* Startup code */
    . = ALIGN(4);
  } >FLASH

  /* The program code and other data goes into FLASH */
  .text :
  {
    . = ALIGN(4);
    *(.text)           /* .text sections (code) */
    *(.text*)          /* .text* sections (code) */
    *(.glue_7)         /* glue arm to thumb code */
    *(.glue_7t)        /* glue thumb to arm code */
    *(.eh_frame)

    KEEP (*(.init))
    KEEP (*(.fini))

    . = ALIGN(4);
    _etext = .;        /* define a global symbols at end of code */
  } >FLASH

  /* Constant data goes into FLASH */
  .rodata :
  {
    . = ALIGN(4);
    *(.rodata)         /* .rodata sections (constants, strings, etc.) */
    *(.rodata*)        /* .rodata* sections (constants, strings, etc.) */
    . = ALIGN(4);
  } >FLASH

  .ARM.extab   : { *(.ARM.extab* .gnu.linkonce.armextab.*) } >FLASH
  .ARM : {
    __exidx_start = .;
    *(.ARM.exidx*)
    __exidx_end = .;
  } >FLASH

  .preinit_array     :
  {
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array*))
    PROVIDE_HIDDEN (__preinit_array_end = .);
  } >FLASH
  .init_array :
  {
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array*))
    PROVIDE_HIDDEN (__init_array_end = .);
  } >FLASH
  .fini_array :
  {
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT(.fini_array.*)))
    KEEP (*(.fini_array*))
    PROVIDE_HIDDEN (__fini_array_end = .);
  } >FLASH

  /* used by the startup to initialize data */
  _sidata = LOADADDR(.data);

  /* Initialized data sections goes into RAM, load LMA copy after code */
  .data : 
  {
    . = ALIGN(4);
    _sdata = .;        /* create a global symbol at data start */
    *(.data)           /* .data sections */
    *(.data*)          /* .data* sections */

    . = ALIGN(4);
    _edata = .;        /* define a global symbol at data end */
  } >SRAM1 AT> FLASH

  _sidtcmram = LOADADDR(.dtcmram);

  /* DTCMRAM section 
  * 
  * IMPORTANT NOTE! 
  * If initialized variables will be placed in this section, 
  * the startup code needs to be modified to copy the init-values.  
  */
  .dtcmram :
  {
    . = ALIGN(4);
    _sdtcmram = .;       /* create a global symbol at dtcmram start */
    *(.dtcmram)
    *(.dtcmram*)

    . = ALIGN(4);
    _edtcmram = .;       /* create a global symbol at dtcmram end */
  } >DTCMRAM AT> FLASH

 _sisram2 = LOADADDR(.sram2);

  /* SRAM2 section 
  * 
  * IMPORTANT NOTE! 
  * If initialized variables will be placed in this section, 
  * the startup code needs to be modified to copy the init-values.  
  */
  .sram2 :
  {
    . = ALIGN(4);
    _ssram2 = .;       /* create a global symbol at sram2 start */
    *(.sram2)
    *(.sram2*)

    . = ALIGN(4);
    _esram2 = .;       /* create a global symbol at sram2 end */
  } >SRAM2 AT> FLASH

  /* Uninitialized data section */
  . = ALIGN(4);
  .bss :
  {
    /* This is used by the startup in order to initialize the .bss secion */
    _sbss = .;         /* define a global symbol at bss start */
    __bss_start__ = _sbss;
    *(.bss)
    *(.bss*)
    *(COMMON)

    . = ALIGN(4);
    _ebss = .;         /* define a global symbol at bss end */
    __bss_end__ = _ebss;
  } >SRAM1

  /* User_heap_stack section, used to check that there is enough SRAM1 left */
  ._user_heap_stack :
  {
    . = ALIGN(4);
    PROVIDE ( end = . );
    PROVIDE ( _end = . );
    . = . + _Min_Heap_Size;
    . = . + _Min_Stack_Size;
    . = ALIGN(4);
  } >SRAM1

  /* MEMORY_bank1 section, code must be located here explicitly            */
  /* Example: extern int foo(void) __attribute__ ((section (".mb1text"))); */
  .memory_b1_text :
  {
    *(.mb1text)        /* .mb1text sections (code) */
    *(.mb1text*)       /* .mb1text* sections (code)  */
    *(.mb1rodata)      /* read-only data (constants) */
    *(.mb1rodata*)
  } >MEMORY_B1

  /* Remove information from the standard libraries */
  /DISCARD/ :
  {
    libc.a ( * )
    libm.a ( * )
    libgcc.a ( * )
  }

  .ARM.attributes 0 : { *(.ARM.attributes) }
}
```
and Here is how I run micropython:
```

  mp_stack_set_top(&_estack);
  mp_stack_set_limit((char *)&_estack - (char *)&_heap_end - 1024);

  // GC init
  //printf("CORE: Starting GC\n");
  gc_init(&_heap_start, &_heap_end);

  // Interpreter init
  //printf("CORE: Starting interpreter\n");
  mp_init();
  mp_obj_list_init(mp_sys_argv, 0);
  mp_obj_list_init(mp_sys_path, 0);
  mp_obj_list_append(
      mp_sys_path,
      MP_OBJ_NEW_QSTR(MP_QSTR_));  // current dir (or base dir of the script)

  // Execute the main script
  //printf("CORE: Executing main script\n");
  pyexec_frozen_module("boot.py");
```

dpgeorge, 2019-08-20:
It's hard to say what the issue is just with the above info.

If you'd like to discuss and ask for help about making a port of MicroPython then the best place to do that is at https://forum.micropython.org

If you think you've found a bug that's reproducible on an existing port/board then please post here with steps to reproduce it.

dpgeorge, 2019-09-27:
Closing, feel free to reopen with reproducible steps for an existing port/board.

closed 2019-09-27 by dpgeorge