micropython/micropython #5030: [STM32] OTP memory usage.

hardkrash, 2019-08-20:
On the STM32 port in commit 370a8116c4348a849c2dafb2c90fa5ba12b8c5dc OTP memory is used to store the MAC address of the pyboard.  I love this!  I intend to store a serial number into the OTP area as well.

Could you share your process used to choose these specific addresses?
I would like to not collide with MicroPython usages vs custom ports.

It appears that the MAC is stored at a specific address, and not based off of FLASH_OTP_BASE nor FLASH_OTP_END.

It appears that this is not FLASH_OTP_END - len(pyb_otp_t) nor a similar offset from FLASH_OTP_BASE

FLASH_OTP_END - OTP_ADDR = 47
len(pyb_otp_t) should equal 10

Am I missing anything?
Would it be a good idea to make OTP addresses or accessor in the STM module?

dpgeorge, 2019-08-21:
> Could you share your process used to choose these specific addresses?

They are `FLASH_OTP_END + 1 - 48`  for F722 and `FLASH_OTP_END + 1 - 96` for F767.

> I would like to not collide with MicroPython usages vs custom ports.

The OTP area is programmed during production and not by any code in this repository.  It's a very board specific thing, and so any custom board would need to define they location and layout of the info in the OTP itself.

The code here to read the OTP should probably be moved to board specific source files, as discussed in #5029.  Then there's nothing to clash with.

jonnor, 2024-09-19:
Proposing to close as out-of-scope.

closed 2025-02-04 by projectgus