micropython/micropython #5031: ujson cannot read byte array or memory view data.

hardkrash, 2019-08-20:
I'm parsing data and want to minimize memory usage but the ujson.loads() only accepts strings and bytes objects.  As shown below.

Is there a memory efficient workaround that I am missing with `ujson.load`?
Considering the small memory footprints of our devices could byte array and memory views be added to the ujson.loads method acceptable input types?

I understand that I can fix this by converting the data to a bytes or string object, but that will create an undesired memory allocation.

Here is an example session showing the issue I am facing.

```python
>>> import json
>>> s = '["hi", "there"]'
>>> b = s.encode('utf-8')
>>> ba = bytearray(b)
>>> mv = memoryview(b)
>>> print(json.loads(s))
['hi', 'there']
>>> print(json.loads(b))
['hi', 'there']
>>> print(json.loads(ba))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'bytearray' object to str implicitly
>>> print(json.loads(mv))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'memoryview' object to str implicitly
```

jimmo, 2019-08-21:
You can use uio.BytesIO to make a stream out of anything supporting the buffer protocol (including bytearray and memoryview).

```
import uio
ujson.load(uio.BytesIO(ba))
```

dpgeorge, 2019-08-21:
> could byte array and memory views be added to the ujson.loads method acceptable input types?

I think this is a good idea, to allow efficient use of memory with `bytearray` (and in MicroPython, if something accepts `str` or `bytes` it usually accepts anything with the buffer protocol, like `bytearray`, `memoryview` and `array.array`).

Actually, CPython does allow to pass in `bytearray` to `json.loads()` (but not `memoryview` it seems...) so it'd make sense to support it from the perspective of CPython compatibility.

dpgeorge, 2019-08-22:
> could byte array and memory views be added to the ujson.loads method acceptable input types?

Done in 2dfa69efbbae92faf21360edd5e60c5a9145a2dc

closed 2019-08-22 by dpgeorge