micropython/micropython #5037: build mpy-cross 'alloca.h' file not found

Needrom, 2019-08-22:
```make``` or ```make V=1``` will raise this error from mine.

make V=1
```
mkdir -p build/genhdr
python3 ../py/makeversionhdr.py build/genhdr/mpversion.h
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
python3 ../py/makemoduledefs.py --vpath="., .., "  py/mpstate.c py/malloc.c py/gc.c py/pystack.c py/qstr.c py/vstr.c py/mpprint.c py/unicode.c py/mpz.c py/reader.c py/lexer.c py/parse.c py/scope.c py/compile.c py/emitcommon.c py/emitbc.c py/asmbase.c py/asmx64.c py/emitnx64.c py/asmx86.c py/emitnx86.c py/asmthumb.c py/emitnthumb.c py/emitinlinethumb.c py/asmarm.c py/emitnarm.c py/asmxtensa.c py/emitnxtensa.c py/emitinlinextensa.c py/formatfloat.c py/parsenumbase.c py/parsenum.c py/emitglue.c py/persistentcode.c py/runtime.c py/runtime_utils.c py/scheduler.c py/nativeglue.c py/stackctrl.c py/argcheck.c py/warning.c py/map.c py/obj.c py/objarray.c py/objattrtuple.c py/objbool.c py/objboundmeth.c py/objcell.c py/objclosure.c py/objcomplex.c py/objdeque.c py/objdict.c py/objenumerate.c py/objexcept.c py/objfilter.c py/objfloat.c py/objfun.c py/objgenerator.c py/objgetitemiter.c py/objint.c py/objint_longlong.c py/objint_mpz.c py/objlist.c py/objmap.c py/objmodule.c py/objobject.c py/objpolyiter.c py/objproperty.c py/objnone.c py/objnamedtuple.c py/objrange.c py/objreversed.c py/objset.c py/objsingleton.c py/objslice.c py/objstr.c py/objstrunicode.c py/objstringio.c py/objtuple.c py/objtype.c py/objzip.c py/opmethods.c py/sequence.c py/stream.c py/binary.c py/builtinimport.c py/builtinevex.c py/builtinhelp.c py/modarray.c py/modbuiltins.c py/modcollections.c py/modgc.c py/modio.c py/modmath.c py/modcmath.c py/modmicropython.c py/modstruct.c py/modsys.c py/moduerrno.c py/modthread.c py/vm.c py/bc.c py/showbc.c py/repl.c py/smallint.c py/frozenmod.c extmod/moductypes.c extmod/modujson.c extmod/modure.c extmod/moduzlib.c extmod/moduheapq.c extmod/modutimeq.c extmod/moduhashlib.c extmod/moducryptolib.c extmod/modubinascii.c extmod/virtpin.c extmod/machine_mem.c extmod/machine_pinbase.c extmod/machine_signal.c extmod/machine_pulse.c extmod/machine_i2c.c extmod/machine_spi.c extmod/modussl_axtls.c extmod/modussl_mbedtls.c extmod/modurandom.c extmod/moduselect.c extmod/moduwebsocket.c extmod/modwebrepl.c extmod/modframebuf.c extmod/vfs.c extmod/vfs_reader.c extmod/vfs_posix.c extmod/vfs_posix_file.c extmod/vfs_fat.c extmod/vfs_fat_diskio.c extmod/vfs_fat_file.c extmod/utime_mphal.c extmod/uos_dupterm.c lib/embed/abort_.c lib/utils/printf.c build/genhdr/moduledefs.h > build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
clang -E -DNO_QSTR -Ibuild/tmp -I. -Ibuild -I.. -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -Os   -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables ../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/pystack.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/reader.c ../py/lexer.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmbase.c ../py/asmx64.c ../py/emitnx64.c ../py/asmx86.c ../py/emitnx86.c ../py/asmthumb.c ../py/emitnthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/emitnarm.c ../py/asmxtensa.c ../py/emitnxtensa.c ../py/emitinlinextensa.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/persistentcode.c ../py/runtime.c ../py/runtime_utils.c ../py/scheduler.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdeque.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/builtinhelp.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../extmod/moductypes.c ../extmod/modujson.c ../extmod/modure.c ../extmod/moduzlib.c ../extmod/moduheapq.c ../extmod/modutimeq.c ../extmod/moduhashlib.c ../extmod/moducryptolib.c ../extmod/modubinascii.c ../extmod/virtpin.c ../extmod/machine_mem.c ../extmod/machine_pinbase.c ../extmod/machine_signal.c ../extmod/machine_pulse.c ../extmod/machine_i2c.c ../extmod/machine_spi.c ../extmod/modussl_axtls.c ../extmod/modussl_mbedtls.c ../extmod/modurandom.c ../extmod/moduselect.c ../extmod/moduwebsocket.c ../extmod/modwebrepl.c ../extmod/modframebuf.c ../extmod/vfs.c ../extmod/vfs_reader.c ../extmod/vfs_posix.c ../extmod/vfs_posix_file.c ../extmod/vfs_fat.c ../extmod/vfs_fat_diskio.c ../extmod/vfs_fat_file.c ../extmod/utime_mphal.c ../extmod/uos_dupterm.c ../lib/embed/abort_.c ../lib/utils/printf.c build/genhdr/moduledefs.h ../py/mpconfig.h mpconfigport.h >build/genhdr/qstr.i.last
In file included from ../py/mpstate.c:27:
In file included from ../py/mpstate.h:31:
In file included from ../py/mpconfig.h:62:
./mpconfigport.h:121:10: fatal error: 'alloca.h' file not found
#include <alloca.h>
         ^~~~~~~~~~
1 error generated.
../py/malloc.c:27:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
../py/gc.c:28:10: fatal error: 'assert.h' file not found
#include <assert.h>
         ^~~~~~~~~~
1 error generated.
../py/pystack.c:27:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
../py/qstr.c:27:10: fatal error: 'assert.h' file not found
#include <assert.h>
         ^~~~~~~~~~
1 error generated.
../py/vstr.c:28:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
../py/mpprint.c:27:10: fatal error: 'assert.h' file not found
#include <assert.h>
         ^~~~~~~~~~
1 error generated.
In file included from ../py/unicode.c:29:
In file included from ../py/unicode.h:29:
In file included from ../py/mpconfig.h:62:
./mpconfigport.h:121:10: fatal error: 'alloca.h' file not found
#include <alloca.h>
         ^~~~~~~~~~
1 error generated.
../py/mpz.c:27:10: fatal error: 'string.h' file not found
#include <string.h>
         ^~~~~~~~~~
1 error generated.
../py/reader.c:27:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
../py/lexer.c:27:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
../py/parse.c:29:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
../py/scope.c:27:10: fatal error: 'assert.h' file not found
#include <assert.h>
         ^~~~~~~~~~
1 error generated.
../py/compile.c:29:10: fatal error: 'stdio.h' file not found
[... 608 lines cut ...]
In file included from ../py/runtime.h:29:
In file included from ../py/mpstate.h:31:
In file included from ../py/mpconfig.h:62:
./mpconfigport.h:121:10: fatal error: 'alloca.h' file not found
#include <alloca.h>
         ^~~~~~~~~~
1 error generated.
In file included from ../lib/utils/printf.c:27:
In file included from ../py/mpconfig.h:62:
./mpconfigport.h:121:10: fatal error: 'alloca.h' file not found
#include <alloca.h>
         ^~~~~~~~~~
1 error generated.
In file included from ../py/mpconfig.h:62:
./mpconfigport.h:121:10: fatal error: 'alloca.h' file not found
#include <alloca.h>
         ^~~~~~~~~~
1 error generated.
mpconfigport.h:121:10: fatal error: 'alloca.h' file not found
#include <alloca.h>
         ^~~~~~~~~~
1 error generated.
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build/genhdr/qstr.i.last'
```

stinos, 2019-08-22:
Can you specify what platform you're compiling on, which version of the source code etc? It's not normal none of the standard headers like stdio.h aren't found so there might be something wrong with your build environment rather than with MicroPython?

Needrom, 2019-08-22:
I close to check out why I miss the stdio.h. I write a .c file with stdio.h which can be compile by gcc. still have no I idea why I missing. I already add the platform and some operational I make.

> Can you specify what platform you're compiling on, which version of the source code etc? It's not normal none of the standard headers like stdio.h aren't found so there might be something wrong with your build environment rather than with MicroPython?

Needrom, 2019-08-22:
it can built if I change the ```CC=clang``` to ```CC=gcc```
my clang will raise the problem like <stdio.h> not found

closed 2019-08-22 by Needrom