micropython/micropython #5040: [esp32]  Micropython runs slower when dual core is started

FFtust, 2019-08-23:
Mp version:  4d94fae83322dd1b4197fd770fa9d0b7474ce72b
esp-idf version: 6b3da6b1882f3b72e904cc90be67e9c4e3f369a9
Hardware：4M SPIRAM + 4M flash
Hi, 
    I found micropython starts slower when dual core is enabled. No matter which core I put micropython on, the phenomenon is the same. It seems that  time consumption in the process of loading modules. I am sorry that I can't show my test code because my system is too complicated.
    Any idear about this problem. Thanks for your help.

dpgeorge, 2019-08-24:
Performance on the the esp32 varies considerably from build-to-build due to different flash layouts interacting with the I-cache.  Also SPIRAM slows things down due to hardware bugs and compiler workarounds (which may not be fully solved yet).  So measuring performance is difficult, and may not be due to single/dual core.  You would need to do lots of performance testing with various build options (eg with and without SPIRAM, slightly varying the firmware, reordering object files within the build, etc) to really pin it down.

FFtust, 2019-08-24:
@dpgeorge 
    Indeed, I need to do more testing to locate this problem, thank you for your suggestions.

closed 2019-08-24 by FFtust

dpgeorge, 2021-10-15:
See also 549448e8bbc8ce0b6b5fc51c0660acdaff18c3d6