micropython/micropython #5043: No LAN.config method on ESP32

paco3346, 2019-08-24:
Looking at the LAN implementation on the ESP32 there's no config method (as seen on other ports such as the STM32. I know micropython doesn't actually provide an abstract class (see http://docs.micropython.org/en/v1.9.3/pyboard/library/network.html#network.config) but this seems like something that would be useful.

Compare https://github.com/micropython/micropython/blob/5801a003f055a9db6d569fbb0ffc3018dcaa8c3e/ports/esp32/network_lan.c#L208
and
https://github.com/micropython/micropython/blob/e61862d063b7a19f5baba8e0d1234d648826a616/ports/stm32/network_lan.c#L153

Specifically, I'm trying to get the mac of a wired interface on an ESP32

MrSurly, 2019-08-24:
Hi @paco3346,

I wrote the LAN port for ESP32.  If your only need is for the MAC address, that can be obtained via the `machine.unique_id()` (Though you might have to increment the last octet to get the LAN MAC -- I'm not sure). Arguably, this isn't portable, and I'm open to modifying `LAN` to include `config`; are there a list of `config` parameters you'd like to see implemented?

@dpgeorge Would a PR for this change be welcome?

paco3346, 2019-08-24:
@MrSurly, thanks for the quick response. For my purposes I only need MAC so the `machine.unique_id()` will work perfectly for me.

I suppose it's possible that in the future I would need more but I can't think of anything else at the moment.

MrSurly, 2019-08-24:
So, just FYI `machine.unique_id()` returns the "base" MAC address, which I _think_ is the Ethernet LAN.  I know that if you take the unique ID and increment the last octet by 1, then it's the WiFi MAC.

dpgeorge, 2019-08-27:
> Would a PR for this change be welcome?

Yes it would be useful to have a `LAN.config()` method, even if only to get the MAC address.

dpgeorge, 2019-08-28:
`LAN.config('mac')` is implemented in 5635b96461e7806c06fccfa237dae670cc2a1377

closed 2019-08-28 by dpgeorge