micropython/micropython #5046: RFC:  MICROPY_PY_BUILTINS_NEXT2 always set to (0) ?

pmp-p, 2019-08-27:
Current settings for all ports break cpython script compatibility on next()

https://docs.python.org/3.5/library/functions.html#next

because
MICROPY_PY_BUILTINS_NEXT2 is never defined to 1 except in coverage and defaults to 0
https://github.com/micropython/micropython/blob/master/py/mpconfig.h#L910

stinos, 2019-08-27:
That choice was made when implementing the feature, to not increase code size with a feature which doesn't get used much, probably because at that point nobody except me requested it.

dpgeorge, 2019-08-30:
For the PR where it was implemented, and associated discussion, see #4400

pmp-p, 2019-08-30:
On unix/javascript/wasi ports where code size does not matter much. compatibility toward *full* 3.4 should be improved at any cost.

 On those platforms MicroPython is judged on its compatibility level while people runing on MCU usually know there are drawbacks. 

 (Useless) lack of compatibility could also be harmfull towards efforts to get a pure Python stdlib alternatives ports are aiming for. 

cf: https://discuss.python.org/t/re-use-of-standard-library-across-implementations/2051/59

stinos, 2019-08-30:
There was a [short discussion ](https://github.com/micropython/micropython/pull/4309#issuecomment-441437495) once on whether unix and maybe other ports should just default to having most/all features enabled (like mpconfigport_coverage.h) vs the situation as it is now.

I don't know how other people use the unix port but I wouldn't be surprised if a majority now use a custom build to get more/all features whereas there are less people who care about having less features/size. Which is indeed another reason to default to as much compatibility as possible for ports running on top of a 'large' OS. But again, just guesswork, I don't know if people really use the unix/windows/javascript ports let alone what they do with it.

pmp-p, 2019-08-30:
> I don't know if people really use the unix/windows/javascript ports let alone what they do with it.

Chicken and egg problem :
  - lack of compat. people won't use it,
  - good compat they'll use it but you can't be sure of that until compatibility is enhanced ( which is harmless if nobody uses it anyway ).

Javascript port is considered for integration in Visual Studio Code to run plugins amonst other solutions.

dpgeorge, 2019-08-30:
> Javascript port is considered for integration in Visual Studio Code to run plugins amonst other solutions.

That's interesting.  Do you have any pointers to online discussion about this?

pmp-p, 2019-08-30:
@dpgeorge  sure  :
### Bundle Python with the extension to run under Node (e.g. JavaScript/WebAssembly implementation)
https://github.com/microsoft/vscode-python/issues/2375

also unrelevant to the issue but probably interesting see the need for per port linter support here https://github.com/microsoft/vscode-python/issues/5098

pmp-p, 2019-08-30:
also please don't mix javascript and wasm-wasi ( even though wasi can run on browser) those are not the same platforms and would have trouble living in same port. so that makes 4 platforms that would need enhanced compatibility.

pmp-p, 2019-09-23:
in continuation of cross posting on another possible RFC  https://github.com/micropython/micropython/issues/5106

> The reason was explained to you less then a month ago in an issue you created specifically for it :)

Background history is always interesting but it does not explain is **why it's still turned OFF almost a month after** at least on ports that don't have any MCU requirements but COMPATIBILITY ones. Strong feelings against it ? need a PR for a couple #define ? too many cooks ? not enough ? let me think of it until `__/__/___` ?  Who knows ....

stinos, 2019-09-23:
Just guesswork, but as I see it dpgeorge labelled this RFC to figure out what others think about whether this should be on by default, plus I guess also how people feel about making the non-MCU ports default to having everything on by default. Nobody replied though, no PR, there's other work to do, so nothing hapened. And in cases like this I tend to agree with that workflow (and the workaround which is just building from source with different switches): we cannot just go enabling things based on 2 people wanting it possibly harming the rest of the community. Or possibly also making the rest happy, but we just don't know. Hence the question for input.