micropython/micropython #5049: [STM32F0] Baudrate of UART(s) is 8 times higher

rolandvs, 2019-08-27:
Tested with the NUCLEO_F091RC port, the UART baudrate is 8 times higher than set.

I need to check some more ( but I need to do some other chores first) but I have set the REPL UART to 8 and the baudrate to 19200 to get REPL working on UART8 with 115200.

First thoughts: maybe the set/get of the baudrate in uart.c for the STM32F0 cases or maybe the clock setting of UART2345678...

rolandvs, 2019-08-28:
Strange

#define  MICROPY_HW_CLK_USE_HSI48 (0) gets the correct result, whereas 
#define  MICROPY_HW_CLK_USE_HSI48 (1) needs the baudrate set to a value  times lower

As far as I can know I use a genuine, non modified NUCLEO_F091RC

Probably staring at me, but I don't see it right now.

dpgeorge, 2019-08-28:
I'm pretty sure that UART2 and UART8 set the correct baudrate, because I use them in a project.

And the NUCLEO_F091RC uses UART2 for the REPL (through the ST-Link USB) at 115200.

rolandvs, 2019-09-10:
@dpgeorge I just tested it again and `#define MICROPY_HW_CLK_USE_HSI48    (0)` works, the default setting does not. Latest MPY, no changes, fresh compile and standard NUCLEO-F091RC.

chrismas9, 2019-09-10:
It appears the F0 ports do not set up the PLL parameters. For the F0 they are different to the other families. There are only two values to set: PREDIV and PLLMUL. For 8MHz crystal and 48MHz SYSCLK these should be set to 1 and 6.

When you disable HSI48 it enables HSE and I suspect the whole system is running at 8MHz (6 times slower). If the baud rate calculation is expecting 48MHz this would explain the error.

Could you enable MCO and measure the SYSCLK frequency using HSE?

rolandvs, 2019-09-10:
The ST-Link on the board provides the 8MHz which is available on `SB50` and connected to the HSE input pin as it did before the HSI48 was introduced :-)

dpgeorge, 2019-09-10:
With the default MPY build UART2 is connected to the ST-link USB-VCP port at 115200, so that UART is at least working.

dpgeorge, 2019-09-10:
All UARTs are by default clocked from PCLK1, which is (should be) 48MHz when HSI48 is enabled.  Doing `machine.freq()` will show PCLK1 as the 3rd entry.  The UART baudrate is calculated from PCLK1 (in the HAL).

So I don't know what is wrong.

chrismas9, 2019-09-10:
I missed where the PLL is initialised because it writes the registers directly and other ports use the HAL. Official F0 ports have always used HSI48 before and after MICROPY_HW_CLK_USE_HSI48 was added. 

Since I am about to use F0 at -40 degrees and need HSE I will investigate further.

chrismas9, 2019-09-13:
I could reproduce the problem and have tested a fix. Buad rate is correct for HSE/PLL and 6 times too fast for HSI48. Using HSI48 I get:
`>>> machine.freq()`
`(48000000, 8000000, 8000000)`

The reason is that `SystemCoreClockUpdate` in stm32lib `system_stm32f0xx.c` does not support HSI48 direct to SYSCLK. In the case statement there are only three cases (HSI, HSE & HSE/PLL). HSI48 falls through to the default which sets `SystemCoreClock = HSI_VALUE;` which is 8 MHz, but the clock is 48 MHz.

The solution I have implemented does not need a change to stm32lib, rather it uses
`HSI48 -> PREDIV (divide by 2) -> PLL (mult by 2) -> SYSCLK`

I will prepare a PR after further testing and checking other F0 and L0 devices.

dpgeorge, 2019-09-26:
Fixed by f16e4be3fa16a3532bb93ccf88e867d27fe2ff20

closed 2019-09-26 by dpgeorge