micropython/micropython #5050: ESP32 - Can't rshell to TinyPICO after using QIO/80MHz in makefile

UnexpectedMaker, 2019-08-28:
I just pulled the latest MP changes from here (thanks for all the work being done atm) and built a new TinyPICO firmware and decided to try QIO and 80MHz this time on my build - as I have moved to that for TinyPICO as the default in Arduino land - but if I build with it, I can’t rshell to it anymore.

I re-built back with DIO and 40MHz and it works - but QIO and 80MHz doesn't. This is definitely supported in IDF land too, so does anyone know why it’s not possible to do this when making MP firmware?

dpgeorge, 2019-08-28:
>  as I have moved to that for TinyPICO as the default in Arduino land

What version (git hash) of the IDF do you use there?

UnexpectedMaker, 2019-08-28:
Is this what you mean?

git log -1
commit 6b3da6b1882f3b72e904cc90be67e9c4e3f369a9 (HEAD, tag: v3.3-beta3)

dpgeorge, 2019-08-28:
No, that's the hash used by this repo.  Which hash does Arduino use, where it's known to work?  It could be that you'll need a more recent version of the IDF than uPy currently uses, or it could be that it's just a configuration setting.

UnexpectedMaker, 2019-08-28:
Oh right, no, Arduino is on an older IDF - v3.2 - hash 6dab3f6777b47e077fecd5b27f5f316664dd6096
https://github.com/espressif/arduino-esp32/releases/tag/1.0.2-rc2

dpgeorge, 2019-08-28:
You may need to add the following to `boards/sdkconfig.spiram`:
```
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
```

UnexpectedMaker, 2019-08-28:
But won't that mean I'm changing the base ESP32 SPIRAM config? Shouldn't this be a TinyPICO specific board change? I can't guarantee every ESP32 has matching PSRAM capabilities, but I want TinyPICO to always have this QIO/80MHz.

dpgeorge, 2019-08-28:
> But won't that mean I'm changing the base ESP32 SPIRAM config?

You're right.  I should have said to create a new `boards/TINYPICO/sdkconfig.board` file with these options, then add that to `boards/TINOPICO/mpconfigboard.mk`.

UnexpectedMaker, 2019-08-28:
I just tried what you suggested (on the base for now) and it didn't work. I still can't connect to the TinyPICO via rshell and if I use screen I find out why - I am seeing constant rebooting with this:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff0018,len:4
load:0xfdecccec,len:-322122018
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
1162 mmu set 00130000, pos 00130000
1162 mmu set 00140000, pos 00140000
ets Jun  8 2016 00:22:57
```

UnexpectedMaker, 2019-08-28:
Just letting you know I still can't get this working... I've cleaned up the board files, moved the changes to be specific to the TINYPICO board config, but I'm at a loss as to where to look next.

dpgeorge, 2019-08-30:
With the following config:
```
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_SPIRAM_SPEED_80M=y
```
I was able to get 80MHz to work:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4936
load:0x40078000,len:10264
load:0x40080400,len:6768
entry 0x400806ec
I (523) psram: This chip is ESP32-PICO
I (523) spiram: Found 64MBit SPI RAM device
I (523) spiram: SPI RAM mode: flash 80m sram 80m
I (526) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (534) cpu_start: Pro cpu up.
I (537) cpu_start: Application information:
I (542) cpu_start: Compile time:     Aug 30 2019 14:43:57
I (548) cpu_start: ELF file SHA256:  0000000000000000...
I (554) cpu_start: ESP-IDF:          v3.3-beta1-694-g6b3da6b18
I (561) cpu_start: Starting app cpu, entry point is 0x40084240
I (553) cpu_start: App cpu up.
I (1046) spiram: SPI SRAM memory test OK
I (1046) heap_init: Initializing. RAM available for dynamic allocation:
I (1047) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1053) heap_init: At 3FFBA5F8 len 00025A08 (150 KiB): DRAM
I (1059) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1066) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1072) heap_init: At 40097168 len 00008E98 (35 KiB): IRAM
I (1078) cpu_start: Pro cpu start user code
I (201) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.11-255-g400a128e1-dirty on 2019-08-30; TinyPICO with ESP32-PICO-D4
Type "help()" for more information.
>>> import micropython
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 4098240, used: 5008, free: 4093232
 No. of 1-blocks: 20, 2-blocks: 6, max blk sz: 264, max free sz: 255818
```
It's still DIO (not QIO) but at least it's a start.

dpgeorge, 2019-08-30:
I could also enable QIO, but not for the very first boot stage.  I added `CONFIG_FLASHMODE_QIO=y` to `boards/TINYPICO/sdkconfig.board` and then `FLASH_FREQ = 80m` to `boards/TINYPICO/mkconfigboard.mk`.  The resulting boot message (with `CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y` to see more output):
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6512
ho 0 tail 12 room 4
load:0x40078000,len:11600
load:0x40080400,len:7632
entry 0x40080744
I (31) boot: ESP-IDF v3.3-beta1-694-g6b3da6b18 2nd stage bootloader
I (31) boot: compile time 15:02:43
I (31) boot: Enabling RNG early entropy source...
I (38) qio_mode: Enabling default flash chip QIO
I (43) boot: SPI Speed      : 80MHz
I (47) boot: SPI Mode       : QIO
I (51) boot: SPI Flash Size : 4MB
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00180000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3a29c (238236) map
I (168) esp_image: segment 1: paddr=0x0004a2c4 vaddr=0x3ffb0000 size=0x032a4 ( 12964) load
I (172) esp_image: segment 2: paddr=0x0004d570 vaddr=0x40080000 size=0x00400 (  1024) load
I (175) esp_image: segment 3: paddr=0x0004d978 vaddr=0x40080400 size=0x02698 (  9880) load
I (187) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xd54d4 (873684) map
I (434) esp_image: segment 5: paddr=0x001254f4 vaddr=0x40082a98 size=0x146d0 ( 83664) load
I (462) esp_image: segment 6: paddr=0x00139bcc vaddr=0x400c0000 size=0x0006c (   108) load
I (462) esp_image: segment 7: paddr=0x00139c40 vaddr=0x50000000 size=0x00808 (  2056) load
I (483) boot: Loaded app from partition at offset 0x10000
I (483) boot: Disabling RNG early entropy source...
I (484) psram: This chip is ESP32-PICO
I (488) spiram: Found 64MBit SPI RAM device
I (493) spiram: SPI RAM mode: flash 80m sram 80m
I (498) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (505) cpu_start: Pro cpu up.
I (509) cpu_start: Application information:
I (514) cpu_start: Compile time:     Aug 30 2019 15:02:50
I (520) cpu_start: ELF file SHA256:  0000000000000000...
I (526) cpu_start: ESP-IDF:          v3.3-beta1-694-g6b3da6b18
I (533) cpu_start: Starting app cpu, entry point is 0x40084240
I (523) cpu_start: App cpu up.
I (1018) spiram: SPI SRAM memory test OK
I (1018) heap_init: Initializing. RAM available for dynamic allocation:
I (1018) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1024) heap_init: At 3FFBA5F8 len 00025A08 (150 KiB): DRAM
I (1031) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1037) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1044) heap_init: At 40097168 len 00008E98 (35 KiB): IRAM
I (1050) cpu_start: Pro cpu start user code
I (172) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.11-255-g400a128e1-dirty on 2019-08-30; TinyPICO with ESP32-PICO-D4
Type "help()" for more information.
>>> 
```
Note the line `mode:DIO, clock div:1` near the start: this means it's in DIO 80MHz mode in the first stage bootloader, but then the second stage bootloader upgrades it to QIO, 80MHz.  Trying to get the first stage bootloader to use QIO lead to the same boot loop seen above.

dpgeorge, 2019-08-30:
See #5054 for a patch that enables QIO and 80MHz.

UnexpectedMaker, 2019-08-30:
w00h00, thanks Damien!

UnexpectedMaker, 2019-09-02:
Working great now, thanks!

closed 2019-09-02 by UnexpectedMaker