micropython/micropython #5059: mpy-cross needs to know sizeof(nlr_buf_t) on target when generating native code

jimmo, 2019-09-02:
The native emitter uses `sizeof(nlr_buf_t)` and `sizeof(code_state_t)` (also `offsetof`), but this will always use the host size, potentially generating the wrong offsets for use in the generated code. (For example: stack space to reserve for registers). This happens every time with x64 unix host (10 register slots in `nlr_buf_t`) generating armv7m (needs 16 register slots), resulting in stack corruption when you call a `@native` function.

Raising this bug to discuss the best way to solve it before going too far into implementation. This is similar to how mpy-cross needs to know about map lookup caching and unicode.

Things that mpy-cross needs to know:
 - `sizeof(nlr_buf_t)` (i.e. number of register slots and whether pystack is enabled)
 - `sizeof(code_state_t)`, in particular what the offset of the `state` member is.

The first one can partially be calculated from `native_arch`, but you also need to know whether soft floating point is enabled. So we could add `_soft` variants of the `armv6` and `armv7m` args to `-march`, or a separate `-msoftfp` flag.

But it looks like we also need flags for:
 - pystack enabled
 - settrace enabled
 - stackless enabled

These need to be encoded in the feature byte (which currently has 3 bits spare)?

Alternatively we could directly pass the size of `nlr_buf_t` and `code_state_t` to mpy-cross, i.e. `-mnlr-size=` and `-mcode-state-offset` ? But this definitely won't fit in the feature byte.

dpgeorge, 2019-09-12:
> The first one can partially be calculated from `native_arch`, but you also need to know whether soft floating point is enabled.

IIRC `MP_NATIVE_ARCH_ARMV7EMSP` and `MP_NATIVE_ARCH_ARMV7EMDP` specify single and double precision FP support (they are currently unused because native code doesn't use any FP instructions).

> But it looks like we also need flags for: pystack enabled, settrace enabled, stackless enabled

Good point, I guess that's all part of the ABI.  For settrace and stackless they aren't supported/needed by native code so one option here would be to define a `mp_native_code_state_t` which only has what native needs, and should not have any optional entries.

dpgeorge, 2019-09-26:
The nlr buf issue was fixed by b596638b9b28975adee4a06a92497b5d9dbba34c.

The other issues still need to be fixed, and #5120 enables tests for it.

dpgeorge, 2022-05-17:
The remaining issue was fixed by 1762990579ff8ddc285e24492626fe40b6f2621d

commit b596638b (2019-09-26): mpy-cross: Set number of registers in nlr_buf_t based on native arch.

Fixes #5059.  Done in collaboration with Jim Mussared.
closed 2022-05-17 by dpgeorge