micropython/micropython #5070: [Discussion] Move ESP32 port to CMake

nevercast, 2019-09-05:
With Espressif moving their ESP-IDF to CMake with major version 4.0 and the new hardware ESP32-S2 and the hardware that will follow only being supported by the CMake toolchain, it may be in our interest in investigating porting the esp32 port to the new CMake tooling.

I've not taken the time yet to get familiar with how the build process is different from GNU Make and I've no estimation of how much work will need to be completed. But I am currently of the opinion that it will need to happen at some point.

This issue is to raise awareness and open discussion.

I have received an engineering sample of the new ESP32-S2 boards and will begin playing with them shortly. I'll add more information as I discover it. 

If anyone else has thoughts, please comment. Especially if you've experience with ESP-IDF 4.x

grbd, 2019-09-07:
Just as a suggestion, perhaps Scons as a build system?
Its basically python and you can single step / debug it within vscode.

With CMake i'd say its a step up from Makefiles but I've never found it to be easy to debug, and some of its existing scripts can be a bit black magic. But a lot of this is personal preference.

nevercast, 2019-09-09:
@grbd The motive for the move is not because I like CMake more than GNU Make, but because Espressif's build system and SDK that is used for the ESP32 port is moving to CMake and GNU Make will not be supported in the future. This issue is to encourage discussion about moving MicroPython's ESP32 port from the old Espressif GNU Make SDK to the new Espressif CMake SDK so that we can support future boards that Espressif produce, such as the upcoming ESP32-S2.

grbd, 2019-09-10:
If the idea is to move just the esp32 port from make to cmake (If I'm wrong then please correct me, I may have misinterpreted that)
You might run into some problems with that approach.
Typically the ports Makefiles tend to be integrated with other common / shared makefiles within micropython
to give an example
```
include ../../py/mkenv.mk
include $(TOP)/py/py.mk
```

So I could see maybe a couple of options
1. try and separate / build the micropython stuff just using make, then afterwards use cmake to build the esp32 port followed by some linking
2. duplicate the existing micropython make build scripts into cmake scripts then call those from the esp32 build

Option 1 might be tricky the way things are all tied together, Option 2 would result in duplication since your no longer limiting to just the port.

jimmo, 2019-09-13:
> I would be interested in your work on this if you have it in a branch.

See PR #5107

nevercast, 2019-09-24:
There is ongoing work to support later versions of the IDF and to support Espressifs upcoming chips. There is no reason to keep this issue open.

closed 2019-09-24 by nevercast