micropython/micropython #5072: ESP32 Port - SDCard Mount - Width parameter with esp32spiram version of firmware

grbd, 2019-09-05:
Hi,
I've raised this as a seperate issue as I think it's different from the other SDCard related problem over at
https://github.com/micropython/micropython/issues/4722

After doing a bit of testing using the ESP-WROVER-KIT 4.1
It looks like if you try to specify the width parameter when using machine.SDCard()
this works fine with the esp32 firmware but breaks under esp32spiram

This works fine with both stock firmwares esp32 and esp32spiram
```
import machine
uos.mount(machine.SDCard(), "/sd")
uos.listdir('/sd')
uos.umount('/sd')
```

This works okay with esp32-20190905-v1.11-274-g06661890d.bin
But breaks with esp32spiram-20190905-v1.11-274-g06661890d.bin
The esp32spiram version of the firmware seems to have an issue with taking a width parameter even if it's the same as the default value (4)
```
import machine
uos.mount(machine.SDCard(width=4), "/sd")
uos.listdir('/sd')
uos.umount('/sd')
```

grbd, 2019-09-05:
Another thing to mention is
with the esp32 version of the firmware I see this after a mount
```
I (11040) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

with esp32spiram I don't see this line, but I think it's just a difference in whatever debug output settings have been set to

The error when specifying width=4 with esp32spiram is

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 259
```

BradChan, 2019-09-17:
I got the same issue.

jonnor, 2024-07-27:
Is this still a problem with the latest releases (MicroPython 1.23+) ?

projectgus, 2025-02-11:
It's been a long time, but both code snippets given in the report work on a WROVER-KIT board with current master.

> with the esp32 version of the firmware I see this after a mount
>
> `I (11040) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0`

Yes, this is a weird quirk of the way MicroPython (mostly) disables ESP-IDF log output.

closed 2025-02-11 by projectgus