micropython/micropython #5073: ESP32 wifi vulnerabilities - may need to update ESP IDF

mattytrentini, 2019-09-06:
There have been [three vulnerabilities identified](https://github.com/Matheus-Garbelini/esp32_esp8266_attacks) for the ESP32 (and ESP8266).

Espressif have apparently already patched these vulnerabilities but we may need to update to a newer version of the ESP-IDF.

More details at [Hackaday](https://hackaday.com/2019/09/05/esp8266-and-esp32-wifi-hacked/).

dpgeorge, 2019-09-06:
I saw that IDF 3.3 was just released (we could easily upgrade to this version).  But do you know which version/commit these vulnerabilities are fixed by?

BradenM, 2019-09-07:
According to [Matheus Garbelini](https://github.com/Matheus-Garbelini) (the author of the exploits) both of the vulnerabilities affecting the ESP32 (CVE-2019-12586 & CVE-2019-12587) can be found in:

> * ESP32-IDF Stable release 3.0 and earlier. Vulnerable until July 15, 2019.
> * ESP32-IDF Development Master [#b68f5b4f](https://github.com/espressif/esp-idf/commit/b68f5b4f8cae3d315a4d82dea5f5a623a18e0c72) and earlier. Vulnerable until May 30, 2019.
> [(source)](https://matheus-garbelini.github.io/home/post/esp32-esp8266-eap-crash/)

The exploit is patched in [#8009320](https://github.com/espressif/esp-idf/commit/8009320fb44abaf8acf8a1e1a38a67fc4c8d458c), meaning the [current hash](https://github.com/micropython/micropython/commit/3ee3995be1c6c461110d7908498777241e08a76a) being used is vulnerable.

dpgeorge, 2019-09-08:
Addressed by 5641aa55dddbdf312a1e8dbb8c3936115683840a

closed 2019-09-08 by dpgeorge