micropython/micropython #5075: [STM32F4DISC] Assembly support broken

lsangild, 2019-09-06:
4691b43c8a245451c1f6eab2261a9433d694ef3e breaks assembly support.

Building STM32F4DISC (`make BOARD=STM32F4DISC`) with attached file (rename to .py) in micropython/ports/stm32/modules/ results in the error

> Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 902, in <module>
    main()
  File "../../tools/mpy-tool.py", line 896, in main
    freeze_mpy(base_qstrs, raw_codes)
  File "../../tools/mpy-tool.py", line 840, in freeze_mpy
    rc.freeze(rc.source_file.str.replace('/', '_')[:-3] + '_')
  File "../../tools/mpy-tool.py", line 445, in freeze
    self.freeze_children(parent_name)
  File "../../tools/mpy-tool.py", line 315, in freeze_children
    rc.freeze(self.escaped_name + '_')
  File "../../tools/mpy-tool.py", line 584, in freeze
    self.freeze_module(self.qstr_links, self.type_sig)
  File "../../tools/mpy-tool.py", line 415, in freeze_module
    print('        .n_state = %u,' % self.prelude[0])
TypeError: %u format: a number is required, not NoneType
make: *** [../../py/mkrules.mk:119: build-STM32F4DISC/frozen_mpy.c] Error 1
make: *** Deleting file 'build-STM32F4DISC/frozen_mpy.c'

[assembly.txt](https://github.com/micropython/micropython/files/3583974/assembly.txt)

dpgeorge, 2019-09-06:
Thanks for the report.  Should be fixed by c69f58e6b97b954d06734797f06feeab4e510855

closed 2019-09-06 by dpgeorge

lsangild, 2019-09-09:
The issue has been resolved. Thank you for the quick response.