micropython/micropython #5079: fatal error: esp32/rom/uart.h: No such file or directory

dongouyang, 2019-09-08:
I followed the instruction to build micropython for esp32. Here is the error that I encountered

make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/pi/esp/esp-idf
Current git hash:   a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea
Supported git hash: 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df
CC /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c
/home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c:14:10: fatal error: esp32/rom/uart.h: No such file or directory
 #include "esp32/rom/uart.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:678: build-GENERIC/bootloader//home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_clock.o] Error 1

Here is my makefile
ESPIDF = /home/pi/esp/esp-idf
BOARD = GENERIC
#PORT = /dev/ttyUSB0
#FLASH_MODE = qio
#FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-

include Makefile

robert-hh, 2019-09-08:
These lines tell the message:
```
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path: /home/pi/esp/esp-idf
Current git hash: a7e8d87d3e5ccc9e5ffcd701a1bac587ba4f43ea
Supported git hash: 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df
```
to fix that, go into the esp-idf directory  /home/pi/esp/esp-idf and issue the commands:
```
git checkout 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df
git submodule update --init
```

dongouyang, 2019-09-10:
Thank you.  Following the instructions, I got rid of the warning.  However, still the same error:

ubuntu:~/esp/micropython/ports/esp32 (master) $ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c
/home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c:14:22: fatal error: rom/uart.h: No such file or directory
compilation terminated.
Makefile:678: recipe for target 'build-GENERIC/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_clock.o' failed
make: *** [build-GENERIC/bootloader/~/esp/esp-idf/components/bootloader_support/src/bootloader_clock.o] Error 1
ubuntu:~/esp/micropython/ports/esp32 (master) $

jimmo, 2019-09-10:
This might be better asked on the forum, but can you share your `makefile` (or `GNUmakefile`). My guess is that you might have something like:

```
ESP_IDF = "~/esp/esp-idf"
```

You need to use $(HOME), because of the way variable expansion works. There's some more info in the README.md.

dongouyang, 2019-09-10:
@jimmo  Thank you.  That fixed the uart.h problem.   I encountered another problem:
**make: *** boards/GENERIC: Is a directory.  Stop.**

here is the content of my makefile

```
ESPIDF = $(HOME)/esp/esp-idf
BOARD = GENERIC
#PORT = /dev/ttyUSB0
#FLASH_MODE = qio
#FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-

include Makefile
```

jimmo, 2019-09-10:
That's exactly what mine looks like.

Can you show how you're invoking make? You should just be running `make` on its own.

The only way I can imagine that you'd get that message is something like `make -f boards/GENERIC`, but you just want to run `make` -- it will automatically use your config in `makefile`.

dongouyang, 2019-09-10:
@jimmo  Thank you for your reply.

I just ran make:

```
pi@dev2:~/esp/micropython/ports/esp32 $ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make: *** boards/GENERIC: Is a directory.  Stop.
pi@dev2:~/esp/micropython/ports/esp32 $
```

jimmo, 2019-09-10:
OK I'm basically just guessing at this point, but this error means that `make` is trying to run or include `$(BOARD_DIR)`.

Is there any chance there's a trailing space on the end of the `BOARD = GENERIC` line in `makefile`?

Have you made any modifications to any files?

dongouyang, 2019-09-11:
@jimmo Thank you for your prompt response. 

I reinstalled everything and also tried the version 1.11.   I keep getting a different error:

```
pi@dev:~/esp/micropython/ports/esp32 $ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_common.c
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: word unexpected (expecting ")")
make: *** [Makefile:678: build-GENERIC/bootloader//home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_common.o] Error 2
pi@dev:~/esp/micropython/ports/esp32 $
```

dpgeorge, 2019-09-11:
It looks like you're build on a RPi?  I don't know if that is supported by the IDF.

As a final thing to do, build with `make V=1` and report the full output of that.

dongouyang, 2019-09-11:
@jimmo  Thank you.  Yes. I am building on Raspberry Pi.   Debian Linux is supported, right?   I can try other OS as well.   What would you suggest?  Ubuntu?

here is the full outpout:
```
pi@dev:~/esp/micropython/ports/esp32 $ make V=1
CC /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_common.c
xtensa-esp32-elf-gcc -std=gnu99 -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -I../../lib/mp-readline -I../../lib/netutils -I../../lib/timeutils -Ibuild-GENERIC -I/home/pi/esp/esp-idf/components/bootloader_support/include -I/home/pi/esp/esp-idf/components/bootloader_support/include_bootloader -I/home/pi/esp/esp-idf/components/console -I/home/pi/esp/esp-idf/components/driver/include -I/home/pi/esp/esp-idf/components/driver/include/driver -I/home/pi/esp/esp-idf/components/nghttp/port/include -I/home/pi/esp/esp-idf/components/nghttp/nghttp2/lib/includes -I/home/pi/esp/esp-idf/components/efuse/include -I/home/pi/esp/esp-idf/components/efuse/esp32/include -I/home/pi/esp/esp-idf/components/esp32/include -I/home/pi/esp/esp-idf/components/espcoredump/include -I/home/pi/esp/esp-idf/components/soc/include -I/home/pi/esp/esp-idf/components/soc/esp32/include -I/home/pi/esp/esp-idf/components/ethernet/include -I/home/pi/esp/esp-idf/components/expat/expat/expat/lib -I/home/pi/esp/esp-idf/components/expat/port/include -I/home/pi/esp/esp-idf/components/heap/include -I/home/pi/esp/esp-idf/components/json/include -I/home/pi/esp/esp-idf/components/json/port/include -I/home/pi/esp/esp-idf/components/log/include -I/home/pi/esp/esp-idf/components/newlib/platform_include -I/home/pi/esp/esp-idf/components/newlib/include -I/home/pi/esp/esp-idf/components/nvs_flash/include -I/home/pi/esp/esp-idf/components/freertos/include -I/home/pi/esp/esp-idf/components/esp_ringbuf/include -I/home/pi/esp/esp-idf/components/esp_event/include -I/home/pi/esp/esp-idf/components/tcpip_adapter/include -I/home/pi/esp/esp-idf/components/lwip/lwip/src/include -I/home/pi/esp/esp-idf/components/lwip/port/esp32/include -I/home/pi/esp/esp-idf/components/lwip/include/apps -I/home/pi/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/pi/esp/esp-idf/components/mbedtls/port/include -I/home/pi/esp/esp-idf/components/mdns/include -I/home/pi/esp/esp-idf/components/mdns/private_include -I/home/pi/esp/esp-idf/components/micro-ecc/micro-ecc -I/home/pi/esp/esp-idf/components/spi_flash/include -I/home/pi/esp/esp-idf/components/ulp/include -I/home/pi/esp/esp-idf/components/vfs/include -I/home/pi/esp/esp-idf/components/xtensa-debug-module/include -I/home/pi/esp/esp-idf/components/wpa_supplicant/include -I/home/pi/esp/esp-idf/components/wpa_supplicant/port/include -I/home/pi/esp/esp-idf/components/ethernet/include -I/home/pi/esp/esp-idf/components/app_trace/include -I/home/pi/esp/esp-idf/components/app_update/include -I/home/pi/esp/esp-idf/components/pthread/include -I/home/pi/esp/esp-idf/components/smartconfig_ack/include -I/home/pi/esp/esp-idf/components/sdmmc/include -DIDF_VER=\"v3.3\" -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_BTREE=1  -Iboards/GENERIC -DBOOTLOADER_BUILD=1 -I/home/pi/esp/esp-idf/components/bootloader_support/include_priv -I/home/pi/esp/esp-idf/components/bootloader_support/include -I/home/pi/esp/esp-idf/components/micro-ecc/micro-ecc -I/home/pi/esp/esp-idf/components/efuse/include -I/home/pi/esp/esp-idf/components/esp32 -Wno-error=format -c -MD -o build-GENERIC/bootloader//home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_common.o /home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_common.c
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: word unexpected (expecting ")")
make: *** [Makefile:678: build-GENERIC/bootloader//home/pi/esp/esp-idf/components/bootloader_support/src/bootloader_common.o] Error 2
pi@dev:~/esp/micropython/ports/esp32 $
```

dpgeorge, 2019-09-11:
There's something wrong with the toolchain.  What is the contents of the file `/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc`?  Can you execute that file on its own?

dongouyang, 2019-09-11:
I think so, not sure how to test it.  Here is the result by just running it.

```
pi@dev:~/esp/xtensa-esp32-elf/bin $ ls -la
total 27836
drwxr-xr-x 2 pi pi    4096 May 24 02:27 .
drwxr-xr-x 8 pi pi    4096 May 24 02:27 ..
-rwxr-xr-x 1 pi pi  963656 May 24 02:27 xtensa-esp32-elf-addr2line
-rwxr-xr-x 2 pi pi  991888 May 24 02:27 xtensa-esp32-elf-ar
-rwxr-xr-x 2 pi pi 1315136 May 24 02:27 xtensa-esp32-elf-as
-rwxr-xr-x 2 pi pi  920432 May 24 02:27 xtensa-esp32-elf-c++
lrwxrwxrwx 1 pi pi      20 May 24 02:20 xtensa-esp32-elf-cc -> xtensa-esp32-elf-gcc
-rwxr-xr-x 1 pi pi  962184 May 24 02:27 xtensa-esp32-elf-c++filt
-rwxr-xr-x 1 pi pi  920432 May 24 02:27 xtensa-esp32-elf-cpp
-rwxr-xr-x 1 pi pi    4681 May 24 01:29 xtensa-esp32-elf-ct-ng.config
-rwxr-xr-x 1 pi pi   28552 May 24 02:27 xtensa-esp32-elf-elfedit
-rwxr-xr-x 2 pi pi  920432 May 24 02:27 xtensa-esp32-elf-g++
-rwxr-xr-x 2 pi pi  916784 May 24 02:27 xtensa-esp32-elf-gcc
-rwxr-xr-x 2 pi pi  916784 May 24 02:27 xtensa-esp32-elf-gcc-8.2.0
-rwxr-xr-x 1 pi pi   25888 May 24 02:27 xtensa-esp32-elf-gcc-ar
-rwxr-xr-x 1 pi pi   25824 May 24 02:27 xtensa-esp32-elf-gcc-nm
-rwxr-xr-x 1 pi pi   25888 May 24 02:27 xtensa-esp32-elf-gcc-ranlib
-rwxr-xr-x 1 pi pi  562544 May 24 02:27 xtensa-esp32-elf-gcov
-rwxr-xr-x 1 pi pi  442280 May 24 02:27 xtensa-esp32-elf-gcov-dump
-rwxr-xr-x 1 pi pi  461480 May 24 02:27 xtensa-esp32-elf-gcov-tool
-rwxr-xr-x 1 pi pi 6298976 May 24 02:27 xtensa-esp32-elf-gdb
-rwxr-xr-x 1 pi pi 1025288 May 24 02:27 xtensa-esp32-elf-gprof
-rwxr-xr-x 4 pi pi 1391128 May 24 02:27 xtensa-esp32-elf-ld
-rwxr-xr-x 4 pi pi 1391128 May 24 02:27 xtensa-esp32-elf-ld.bfd
-rwxr-xr-x 2 pi pi  975752 May 24 02:27 xtensa-esp32-elf-nm
-rwxr-xr-x 2 pi pi 1099080 May 24 02:27 xtensa-esp32-elf-objcopy
-rwxr-xr-x 2 pi pi 1259824 May 24 02:27 xtensa-esp32-elf-objdump
-rwxr-xr-x 2 pi pi  991888 May 24 02:27 xtensa-esp32-elf-ranlib
-rwxr-xr-x 2 pi pi  573080 May 24 02:27 xtensa-esp32-elf-readelf
-rwxr-xr-x 1 pi pi  964552 May 24 02:27 xtensa-esp32-elf-size
-rwxr-xr-x 1 pi pi  963976 May 24 02:27 xtensa-esp32-elf-strings
-rwxr-xr-x 2 pi pi 1099080 May 24 02:27 xtensa-esp32-elf-strip
pi@dev:~/esp/xtensa-esp32-elf/bin $ /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-bash: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: cannot execute binary file: Exec format error
pi@dev:~/esp/xtensa-esp32-elf/bin $
```

dongouyang, 2019-09-11:
Determined to get it work, I got an AWS Ubuntu instance, I got much further. however, still ran into a problem at the end.  

```
ubuntu@ip-172-31-25-1:~/esp/micropython/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_clock.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_common.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_init.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/bootloader_utility.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/secure_boot_signatures.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/secure_boot.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/esp_image_format.c
CC /home/ubuntu/esp/esp-idf/components/bootloader_support/src/flash_encrypt.c
AR build-GENERIC/bootloader/libbootloader_support.a
CC /home/ubuntu/esp/esp-idf/components/spi_flash/spi_flash_rom_patch.c
AR build-GENERIC/bootloader/libspi_flash.a
CC /home/ubuntu/esp/esp-idf/components/soc/esp32/rtc_clk.c
CC /home/ubuntu/esp/esp-idf/components/soc/esp32/rtc_wdt.c
AR build-GENERIC/bootloader/libsoc.a
LINK build-GENERIC/bootloader.elf
Create build-GENERIC/bootloader.bin
esptool.py v2.8-dev
CC ../../py/mpstate.c
CC ../../py/nlr.c
CC ../../py/nlrx86.c
CC ../../py/nlrx64.c
CC ../../py/nlrthumb.c
CC ../../py/nlrxtensa.c
CC ../../py/nlrsetjmp.c
CC ../../py/malloc.c
CC ../../py/gc.c
CC ../../py/pystack.c
CC ../../py/qstr.c
CC ../../py/vstr.c
CC ../../py/mpprint.c
CC ../../py/unicode.c
CC ../../py/mpz.c
CC ../../py/reader.c
CC ../../py/lexer.c
CC ../../py/parse.c
CC ../../py/scope.c
CC ../../py/compile.c
CC ../../py/emitcommon.c
CC ../../py/emitbc.c
CC ../../py/asmbase.c
CC ../../py/asmx64.c
CC ../../py/emitnx64.c
CC ../../py/asmx86.c
CC ../../py/emitnx86.c
CC ../../py/asmthumb.c
CC ../../py/emitnthumb.c
CC ../../py/emitinlinethumb.c
CC ../../py/asmarm.c
CC ../../py/emitnarm.c
CC ../../py/asmxtensa.c
CC ../../py/emitnxtensa.c
CC ../../py/emitinlinextensa.c
CC ../../py/formatfloat.c
CC ../../py/parsenumbase.c
CC ../../py/parsenum.c
CC ../../py/emitglue.c
CC ../../py/persistentcode.c
CC ../../py/runtime.c
CC ../../py/runtime_utils.c
CC ../../py/scheduler.c
CC ../../py/nativeglue.c
CC ../../py/stackctrl.c
CC ../../py/argcheck.c
CC ../../py/warning.c
[... 193 lines cut ...]
                            ^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:34:16: note: in expansion of macro 'LAST_FREE_BIT_BLK2'
 _Static_assert(LAST_FREE_BIT_BLK2 <= MAX_BLK_LEN, "The eFuse table does not match the coding scheme. Edit the ta                ^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:31:28: error: expression in static assertion is not an integer
 #define LAST_FREE_BIT_BLK3 192
                            ^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:35:16: note: in expansion of macro 'LAST_FREE_BIT_BLK3'
 _Static_assert(LAST_FREE_BIT_BLK3 <= MAX_BLK_LEN, "The eFuse table does not match the coding scheme. Edit the ta                ^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: error: initializer element is not constant
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:71:21: note: in expansion of macro 'MAX_BLK_LE
'
     {EFUSE_BLK1, 0, MAX_BLK_LEN},   // Flash encrypt. Key. (length = "None" - 256. "3/4" - 192. "REPEAT" - 128),
^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:26:21: note: (near initialization for 'ENCRYPT_FLASH_KEY[0].bit_count')
 #define MAX_BLK_LEN CONFIG_EFUSE_MAX_BLK_LEN
                     ^
/home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.c:71:21: note: in expansion of macro 'MAX_BLK_LE
'
     {EFUSE_BLK1, 0, MAX_BLK_LEN},   // Flash encrypt. Key. (length = "None" - 256. "3/4" - 192. "REPEAT" - 128),
^
../../py/mkrules.mk:47: recipe for target 'build-GENERIC//home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.o' failed
make: *** [build-GENERIC//home/ubuntu/esp/esp-idf/components/efuse/esp32/esp_efuse_table.o] Error 1
```

dongouyang, 2019-09-11:
Thank you @jimmo @dpgeorge 
I start a different version of Ubuntu in AWS and it worked! Two last questions:

1. I want to change boot.py and add main.py to the image.  Where do I add them?
2. There are several .bin files.  Do I flash firmware.bin to the board?

closed 2019-09-15 by dongouyang