micropython/micropython #5080: ESP 32 cam ,, micropython Won't deploy

Jasou, 2019-09-08:
$ make deploy
CC /home/pi/micropython/esp-idf/components/soc/esp32/rtc_clk.c
make: xtensa-esp32-elf-gcc: Command not found
make: *** [Makefile:991: build/bootloader//home/pi/micropython/esp-idf/components/soc/esp32/rtc_clk.o] Error 127

dpgeorge, 2019-09-09:
It looks like you need to install the xtensa toolchain (compiler).  See `ports/esp32/README.md` for details.

Jasou, 2019-09-09:
Thank you dpgeorge  but i got this error now : 
 CC /home/pi/micropython/esp-idf/components/soc/esp32/rtc_clk.c
/home/pi/micropython/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/micropython/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
make: *** [Makefile:991: build/bootloader//home/pi/micropython/esp-idf/components/soc/esp32/rtc_clk.o] Error 2

jonnor, 2024-07-27:
There is not information here to be able to debug this issue. The ESP32 build works in general. Proposing to close.

robert-hh, 2024-07-27:
Most likely resolved.

closed 2024-07-27 by robert-hh