micropython/micropython #5087: [ESP32] CAN bus and MOD bus master RTU.

beyonlo, 2019-09-09:
Hello.

1. Are there support or plans to support CAN bus on ESP32? I see that there are a implementation for pyboard https://docs.micropython.org/en/latest/library/pyb.CAN.html  - is that a similar implementation on the ESP32?

2. And about the MOD bus master RTU on micropython?
I think that, differently of CAN bus, MOD bus implementation is not a microcontroller dependent, just works over UART, is that correct? 
I not found a implementation for the modbus master RTU in micropython libs and so on, anyone know if are there a third lib for that?

Thank you so much.

rolandvs, 2019-09-11:
There has been some discussion on the Micropython forum about modbus:
```https://forum.micropython.org/viewtopic.php?t=5453```

nos86, 2019-09-23:
Hello,
I'm interested on CAN communication as well.
I know that Pumbaa (a fork of Micropython) exists, but its creator said me it is no longer mantained.

nos86, 2019-10-06:
Hello,
I just give an update to everyone is interested.
Since I was not able to find nothing on the web, I've decided to write the module by myself.
A very first version has been published on my repository. If anyone is interested, here the link of it: https://github.com/nos86/micropython/tree/esp32-can-driver-v1
Since the branch is in continuous development, I prefer to give you the link to the tag of first stable version.

Do not hesitate to open an issue for any new request or just to highlight any bug on the code.

coffice12, 2019-10-07:
> Hello,
> I just give an update to everyone is interested.
> Since I was not able to find nothing on the web, I've decided to write the module by myself.
> A very first version has been published on my [...]

hava a example?

nos86, 2019-10-09:
I have updated the Quick Ref file. Here the link to CAN Bus section: https://github.com/nos86/micropython/blob/esp32-can-driver-v1/docs/esp32/quickref.rst#can-bus

beyonlo, 2019-10-09:
@nos86 

Great work. Maybe you need to contact the @dpgeorge to commit this CAN Bus driver on the master libs. :)

By the way, did you wrote or something with a MOD bus RTU Master on MicroPython? I need to use that to read/write from ESP32 to a slave devices that implement MOD bus RTU.

Thank you.

nos86, 2019-10-10:
Actually not, my goal was to develop the driver for CAN hardware for my project. 
Regarding the merge in master, I'd like to add some additional basic function before request a pull.

camosoul, 2019-11-07:
Things go slow when not incentivized...

Might a modest bounty speed things up? I'd really like a proper CAN implementation (with proper documentation to go with it) for ESP32...

I can't code worth a damn, but I can spend money...

mlokhandwala, 2019-12-12:
Hi @nos86,
So how does one install the CAN bus driver? Is it install-able by pip or does one have to compile the source code? I am desperately looking to use the CAN bus in my project and currently having to make do using the UART although the PCB is intended to communicate via CAN.

camosoul, 2019-12-12:
> Hi @nos86,
> So how does one install the CAN bus driver? Is it install-able by pip or does one have to compile the source code? I am desperately looking to use the CAN bus in my project and currently having to make do using the UART although the PCB is intended to communicate via CAN.

It's about to be a native part of MicroPython thanks  to this guy.

neographikal, 2019-12-15:
Great work! I really could use this for a project I'm doing since I don't care much for writing all this C-code. How does the ESP32 hold up with CAN & micropython when the can bus is quite busy?

nos86, 2019-12-17:
Hello @neographikal,
for the first release, I'm not sure I'm able to include the configuration of the filter.
If so, the software filtering at uPython level is not suitable for high bus-load conditions.

For the next releases, I'm planning to incorporate the configuration of hardware filter (best option for very high bus load) and some software filter handled at low level by C in order to improve the flexibility of the library.

neographikal, 2019-12-17:
Good to know :) Thanks!

neographikal, 2019-12-23:
> 
> 
> I have updated the Quick Ref file. Here the link to CAN Bus section: https://github.com/nos86/micropython/blob/esp32-can-driver-v1/docs/esp32/quickref.rst#can-bus

Going to ask some stupid questions;

Is this lib meant for the onboard can bus controller on the ESP32? The SDA/SDL references threw me off a bit, or is the internal one also i2c based? I'm using the internal controller with an 2551 transceiver.

I have an proof of concept working in C now;

```
void setup() {
  Serial.begin(115200);
  CAN_cfg.speed = CAN_SPEED_250KBPS;
  CAN_cfg.tx_pin_id = GPIO_NUM_5;
  CAN_cfg.rx_pin_id = GPIO_NUM_4;
  CAN_cfg.rx_queue = xQueueCreate(rx_queue_size, sizeof(CAN_frame_t));
  // Init CAN Module
  ESP32Can.CANInit();
}
```
is the baudrate parameter 250 for 250kbps?

Thanks :) Going to try this, nice to have some spare time in the holidays ;)

nos86, 2019-12-23:
Driver is designed for internal controller with external transceiver (likes mcp2551).

Please find the example inside my foto (example/esp32_can.py) https://github.com/nos86/micropython/blob/esp32-can-driver/examples/esp32_can.py

There you can find home to initialize the driver. More details on https://github.com/nos86/micropython/blob/esp32-can-driver/docs/library/esp32.CAN.rst

riegaz, 2019-12-26:
@nos86  Thanks a lot. I'm really looking forward that you CAN driver is included into Micropython. Do you have a build ready to be tested?

Another question I have is: What about sniffing a CAN bus that has both, standard and extended frames. Would it work? Is it possible to send both types on the same bus one after the other?

riegaz, 2019-12-26:
Can there be two CAN bus at the same time? What are the available pin combinations?

nos86, 2019-12-27:
According to my knowledge, esp32 has only one can driver.
Currently I don't have a build available on GitHub, but you can build it from source.

I didn't try to use mixed messages, I'll try it as soon as a return from vacation.
In any case it depends on the firmware implementation at esp-idf level. No limitations are present at micropython level

mlokhandwala, 2020-01-02:
Hello @nos86 can you release a build? You have forked from a reasonably new version of repository and in fact I (possibly many others) are using an older MicroPython release.  So we could avoid setting up the whole build system if you could make a recent build available.

nos86, 2020-01-02:
I'm on vacation without internet. I'll be able to upload a build by end of next week.

nos86, 2020-01-16:
Hi @jbakuwel,
you can download my last nightly build at https://github.com/nos86/micropython/releases/download/esp32-can-driver-v2/firmware.bin

jbakuwel, 2020-01-17:
Hi @nos86,
Firmware loaded on the board successfully. Many thanks for your help!

mlokhandwala, 2020-01-17:
Hi @nos86 @jbakuwel 

Sorry for this noobish question but how can I integrate the firmware.bin with the Micropython installation on the ESP32? Is this to be copied to a specific location there? How will the API get integrated and exposed?

nos86, 2020-01-17:
@mlokhandwala, firmware.bin is the hex file to be loaded in ESP32. It is a Mycropython installation that includes the CAN driver. Please refer to https://docs.micropython.org/en/latest/esp32/tutorial/intro.html#deploying-the-firmware for the installation. Clearly, you have to use my firmware instead the official one.

camosoul, 2020-01-17:
> Hi @nos86 @jbakuwel
> 
> Sorry for this noobish question but how can I integrate the firmware.bin with the Micropython installation on the ESP32? Is this to be copied to a specific location there? How will the API get integrated and exposed?

`esptool.py --chip esp32 --port /dev/ttyUSB0 erase_flash`

`curl -LO https://github.com/nos86/micropython/releases/download/esp32-can-driver-v2/firmware.bin`

`esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 write_flash -z 0x1000 firmware.bin`

mlokhandwala, 2020-01-18:
Hi @nos86 

> @mlokhandwala, firmware.bin is the hex file to be loaded in ESP32. It is a Mycropython installation that includes the CAN driver. Please refer to [...]

My apologies! I was expecting it to be the full installation but when I saw the size of the firmware.bin for some reason I thought it was too small and probably just the CAN lib.

mlokhandwala, 2020-01-29:
Hi @nos86 ,

I suppose the build you have uploaded is for modules with SPI. After I flash it I get the following error (loop):

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee

I am using a chip with the following module:

Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 160MHz,
None
Crystal is 40MHz
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB

nos86, 2020-02-01:
I made no changes in Makefile. it is the same of the mainstream.
I developed on my ESP-WROOM-32 board. I'm not sure what could be the problem in you case.

Have you tried to flash again the official micropython firmware to check if it works?

mlokhandwala, 2020-02-01:
Hi, 
The official MicroPython firmware works perfectly. 

I guess there must be some make file switch for SPI / No SPI Flash. Even the official firmware is available in SPI / No SPI Flash versions.

skpang, 2020-02-19:
Hi,
I've just flash the firmware with the date 2020-01-0 and tried the example but getting this error:

MicroPython esp32-can-driver-v2-4-gc8afef6e8-dirty on 2020-01-09; ESP32 module with ESP32
Type "help()" for more information.
>>> from machine import CAN
>>> bus = CAN(tx=4, rx=2, baudrate=500, mode=CAN.MODE_NO_ACK)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: type object 'CAN' has no attribute 'MODE_NO_ACK'

Any idea of how to fix this error ?

nos86, 2020-02-20:
Hello skpang,
that constant has been changed to CAN.SILENT
Please refer to docs/esp32/quickref.rst

Can you try it and tell me if this fixes your problem?

skpang, 2020-02-20:
CAN.SILENT didn't work either.
>>> bus = CAN(tx=4, rx=2, baudrate=500, mode=CAN.SILENT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 1 required positional arguments

I am looking at the example at:
https://github.com/nos86/micropython/blob/esp32-can-driver-v1/docs/esp32/quickref.rst

Also at:
https://github.com/nos86/micropython/blob/esp32-can-driver-v1/ports/esp32/machine_can.c
There doesn't seems to be a SILENT defined anywhere.

skpang, 2020-02-20:
I also tried:
>>> bus = CAN(0, extframe=True, mode=CAN.LOOPBACK, baudrate=CAN.BAUDRATE_500k)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: type object 'CAN' has no attribute 'BAUDRATE_500k'

Is there a newer binary image available for the ESP32 with CAN ?

nos86, 2020-02-22:
Please, download latest version: https://github.com/nos86/micropython/releases/tag/esp32-can-driver-v3

Refer to https://github.com/nos86/micropython/blob/esp32-can-driver-v3/docs/library/machine.CAN.rst as documentation for usage.

Several changes has been done since first version in order to harmonize the driver across different hardwares.

Moreover, pay attention, in order to save space, several constants have been removed from code. If you want to use it, you have to declare in your script.

skpang, 2020-02-22:
Hi nos86,

I've just tried the new v3 binary but after flashed the board I'm getting this error:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:228892

Is this the correct binary for an Adafruit HUZZAH32 – ESP32 Feather Board ?

Tbruno25, 2020-02-28:
@nos86 I'm running into the same issue as @skpang  on 2 different versions of esp32 dev boards. Just goes into an infinite error loop.

Civilduino, 2020-03-13:
Hello @nos86, first great work implementing the CAN drivers. Did you finish the complete implementation with filters? How can I use it with the lastest MicroPython and not your bin file?

Tbruno25, 2020-06-06:
@nos86 Anything I can do to help with the development for this?

jbakuwel, 2020-06-22:
@nos86, @Tbruno25 for what it's worth: [v2](https://github.com/nos86/micropython/releases/download/esp32-can-driver-v2/firmware.bin) of the firmware loads fine on the ESP32 [board](https://www.openvehicles.com/developers) I have here but v3 goes into the mentioned infinite boot loop. What has changed between v2 and v3?

@nos86, The board I have has three CAN buses, do you think it's feasible to make all those accessible with uPython?

Tbruno25, 2020-10-11:
While we are waiting for this to be added to the official micropython repo, if you need CAN bus you can give [this firmware](https://github.com/Tbruno25/pycom-esp32-universal) a try. It's a modified version of Pycom's WIPY firmware which implements CAN -- confirmed working with a sn65hvd230 on an esp32-wroom-32.

hklang10, 2020-11-30:
I have forked for a personal project, it is work in progress. 

https://github.com/hklang10/micropython/tree/esp32-can-updates

This should build against micropython/master

newbie-gk, 2021-02-23:
I see that it's been a while since the last update....
Any updates on the request being implemented in the official micropython?

IhorNehrutsa, 2021-06-22:
WIP: esp32 CAN(TWAI) driver
https://github.com/micropython/micropython/pull/7381

FeroxTL, 2022-09-21:
Is there any progress with this PR?

I was able to build it with latest master version (1.19.1) with replacing definition of `const mp_obj_type_t machine_can_type` to 

```cpp
MP_DEFINE_CONST_OBJ_TYPE(
    machine_can_type,
    MP_QSTR_CAN,
    MP_TYPE_FLAG_NONE,
    make_new, machine_hw_can_make_new,
    print, machine_hw_can_print,
    locals_dict, (mp_obj_dict_t *)&machine_can_locals_dict
);
```

but it is not working with my hardware. Maybe I missed something? I'm using VP230 for converting TTL levels of esp32 to CAN BUS levels. According to oscilloscope it should send and receive some data, but the only thing I can see is increasing `rx_error_counter` and `bus_error_count` in `can.info()`