micropython/micropython #5089: ESP32-D2WD

tcpipchip, 2019-09-10:
[image]

This new patch is making my NINA W102 reset

tcpipchip, 2019-09-10:
A new step by step should be good for all!

tcpipchip, 2019-09-10:
i had to change partitions-2MiB.csv to
nvs,        data, nvs,     0x9000,   0x6000,
phy_init,   data, phy,     0xf000,   0x1000,
factory,    app,  factory, 0x10000,  0x180000,
vfs,        data, fat,     0x190000, 0x20000,

and makefile to 
##REPAIR

ESPIDF = /home/osboxes/Micropython/esp32/esp-idf
#PORT = /dev/ttyUSB0
#FLASH_MODE = qio
#FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-

##include Makefile

PORT = /dev/ttyUSB0
FLASH_MODE = dio
FLASH_FREQ = 40m
## if using a 2MiB ESP32 board
FLASH_SIZE = 2MB
FLASH_USER_START = 0x120000 # Defaults to 0x200000 address if not provided
FLASH_STORAGE_PARTITION_SIZE = 0xA0000 # 640MiB max. Defaults to 0x200000 if not provided (2MiB partiti$
PART_SRC = partitions-2MiB.csv # Use the custom partitions CSV for 2MiB devices

include Makefile

and modesp.c
to 
STATIC mp_obj_t esp_flash_user_start(void) {
    return MP_OBJ_NEW_SMALL_INT(0x190000);
}

To work!

tcpipchip, 2019-09-10:
Tomorrow wil try again mqtt with 3 certificates!

dpgeorge, 2019-09-11:
@zrecore I think the `partitions-2MiB.csv` is not good because the size available for the main firmware is 0x110000 which is too small by about 10k for the current image of around 1200k.

Also it'd be good to expand the vfs to take maximum possible free space.

I'd suggest that `partitions-2MiB.csv` be changed to:
```
nvs,        data, nvs,     0x9000,   0x6000,
phy_init,   data, phy,     0xf000,   0x1000,
factory,    app,  factory, 0x10000,  0x1b0000,
vfs,        data, fat,     0x1c0000, 0x40000,
```

@tcpipchip does the above change fix it for you?

tcpipchip, 2019-09-11:
Yes, works too!

I changed Makefile to GENERIC_DWD2 and works too!

dpgeorge, 2019-09-11:
@tcpipchip you must be on another branch because neither `FLASH_USER_START` nor `FLASH_STORAGE_PARTITION_SIZE` exist in the master branch.  Try using master with the changes to partitions-2MiB.csv.

tcpipchip, 2019-09-11:
do you mean that i can remove 
FLASH_USER_START and FLASH_STORAGE_PARTITION_SIZE from **m**akefile ?

dpgeorge, 2019-09-11:
Yes.

tcpipchip, 2019-09-11:
following the partition info
nvs,        data, nvs,     0x9000,   0x6000,
phy_init,   data, phy,     0xf000,   0x1000,
factory,    app,  factory, 0x10000,  0x1b0000,
vfs,        data, fat,     0x1c0000, 0x40000,

i changed in modesp.c to
STATIC mp_obj_t esp_flash_user_start(void) {
    return MP_OBJ_NEW_SMALL_INT(0x1c0000);
}
and **m**akefile has
##REPAIR

ESPIDF = /home/osboxes/Micropython/esp32/esp-idf
#PORT = /dev/ttyUSB0
#FLASH_MODE = qio
#FLASH_SIZE = 4MB
#CROSS_COMPILE = xtensa-esp32-elf-

##include Makefile

PORT = /dev/ttyUSB0
FLASH_MODE = dio
FLASH_FREQ = 40m
## if using a 2MiB ESP32 board
FLASH_SIZE = 2MB
#FLASH_USER_START = 0x120000 # Defaults to 0x200000 address if not provided
#FLASH_STORAGE_PARTITION_SIZE = 0xA0000 # 640MiB max. Defaults to 0x200000 if not provided (2MiB partit$
#PART_SRC = partitions-2MiB.csv # Use the custom partitions CSV for 2MiB devices

include Makefile

and **M**akefile has

# Select the board to build for: if not given on the command line,
# then default to GENERIC.
BOARD ?= GENERIC_D2WD
=================
It`s working
Will the actual GENERIC_D2WD patch to uses that configs in those files ?