micropython/micropython #5091: qstr.i.last does not depend on SDKCONFIG_H

nevercast, 2019-09-11:
https://github.com/micropython/micropython/blob/862cc45a9c3241101c5d89e18063724793f73e43/py/mkrules.mk#L73

It seems there is a race condition because qstr.i.last does not depend on sdkconfig.h being present.

```
GEN build-node/sdkconfig.h
GEN build-node/genhdr/mpversion.h
GEN build-node/genhdr/moduledefs.h
GEN build-node/genhdr/qstr.i.last
In file included from /builder/esp-idf/components/esp32/include/esp_assert.h:17:0,
                 from /builder/esp-idf/components/soc/esp32/include/soc/soc.h:20,
                 from /builder/esp-idf/components/esp32/include/rom/ets_sys.h:21,
                 from ./mpconfigport.h:9,
                 from ../../py/mpconfig.h:62,
                 from ../../py/obj.h:29,
                 from /user_c_modules/deepsleep_stub/mod_dsstub.c:2:
/builder/esp-idf/components/newlib/platform_include/assert.h:20:23: fatal error: sdkconfig.h: No such file or directory
...
../../py/mkrules.mk:74: recipe for target 'build-node/genhdr/qstr.i.last' failed
make: *** [build-node/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-node/genhdr/qstr.i.last'
make: *** Waiting for unfinished jobs....
```

I am using `make -j5`, Its my assumption that running make in parallel allows this issue to occur. I have also only ever seen this issue with User C Modules, but not always the same module.

nevercast, 2019-09-11:
I can `repro.sh` this consistently on my end with 

```bash
#!/bin/bash

# Remove SDKCONFIG_H and qstr.i.last
rm micropython/ports/esp32/build-hub/sdkconfig.h
rm micropython/ports/esp32/build-hub/genhdr/qstr.i.last

# Prepare source files in correct directories
# Copies files to ./bundle from ./src
./build.sh bundle

# Run make with correct environment paths
# Equal to:
# make -j5 DEBUG=1 RUNDIR="$RUNDIR" BOARD=hub BOARD_DIR=$(realpath ./src/boards/hub) FROZEN_MPY_DIR=$(realpath ./bundle/hub/src) USER_C_MODULES=$(realpath ./bundle/hub/modules)
./build.sh local-hub
```

Unfortunately I'm not sure I can show my ./build.sh, so I've explained what it does equivilently. 

*Reproduce:*
* Run make
* Success
* Delete sdkconfig.h and qstr.i.last
* Run make
* Fails

Fortunately it doesn't always fail with User C Modules, as I managed to prove with my `repro.sh`

```
In file included from /home/josh/Desktop/esp32/upy/lora/esp-idf/components/esp32/include/esp_assert.h:17:0,
                 from /home/josh/Desktop/esp32/upy/lora/esp-idf/components/soc/esp32/include/soc/soc.h:20,
                 from /home/josh/Desktop/esp32/upy/lora/esp-idf/components/esp32/include/rom/ets_sys.h:21,
                 from ./mpconfigport.h:9,
                 from ../../py/mpconfig.h:62,
                 from ../../py/mpstate.h:31,
                 from ../../py/runtime.h:29,
                 from ../../extmod/modbtree.c:32:
/home/josh/Desktop/esp32/upy/lora/esp-idf/components/newlib/platform_include/assert.h:20:23: fatal error: sdkconfig.h: No such file or directory
```

nevercast, 2019-09-11:
Better repro:

```
cd /home/josh/Desktop/esp32/upy/lora/micropython/ports/esp32
export ESPIDF=/home/josh/Desktop/esp32/upy/lora/esp-idf/
export PATH=$PATH:/home/josh/Desktop/esp32/xtensa/linux64-1.22.0-80-g6c4433a/bin/
make clean
make -j5 
rm build-GENERIC/sdkconfig.h
rm build-GENERIC/genhdr/qstr.i.last
make -j5
```

This should work on latest MicroPython and have nothing to do with my setup.

nevercast, 2019-09-11:
When I change `mkrules.mk:73..75` from 
```
$(HEADER_BUILD)/qstr.i.last: $(SRC_QSTR) $(QSTR_GLOBAL_DEPENDENCIES) | $(HEADER_BUILD)/mpversion.h
	$(ECHO) "GEN $@"
	$(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $(if $(filter $?,$(QSTR_GLOBAL_DEPENDENCIES)),$^,$(if $?,$?,$^)) >$(HEADER_BUILD)/qstr.i.last
```
to
```
$(HEADER_BUILD)/qstr.i.last: $(SDKCONFIG_H) $(SRC_QSTR) $(QSTR_GLOBAL_DEPENDENCIES) | $(HEADER_BUILD)/mpversion.h
	$(ECHO) "GEN $@"
	$(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $(if $(filter $?,$(QSTR_GLOBAL_DEPENDENCIES)),$^,$(if $?,$?,$^)) >$(HEADER_BUILD)/qstr.i.last
```
I'm given another issue: 
```
build-GENERIC/sdkconfig.h:5:9: error: #pragma once in main file [-Werror]
 #pragma once
         ^
cc1: all warnings being treated as errors
```
Derp

nevercast, 2019-09-11:
Changing `mkrules.mk:73..75` to 
```
$(HEADER_BUILD)/qstr.i.last: $(SRC_QSTR) $(QSTR_GLOBAL_DEPENDENCIES) | $(HEADER_BUILD)/mpversion.h $(SDKCONFIG_H)
	$(ECHO) "GEN $@"
	$(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $(if $(filter $?,$(QSTR_GLOBAL_DEPENDENCIES)),$^,$(if $?,$?,$^)) >$(HEADER_BUILD)/qstr.i.last
```

Seems to have fixed it, helps to take the time to understand a target

dpgeorge, 2019-09-11:
In order not to change the common `py/mkrules.mk` does it work to add the following dependency to the esp32 Makefile?
```
$(HEADER_BUILD)/mpversion.h: $(SDKCONFIG_H)
```

nevercast, 2019-09-11:
> In order not to change the common `py/mkrules.mk` does it work to add the following dependency to the esp32 Makefile?

I was just checking that avenue, wasn't sure if `SDKCONFIG_H` already depended on `mpversion.h`.

I've just added your suggested line verbatim and it seems to have worked.

`ports/esp32/Makefile:289`
```
@ $(HEADER_BUILD)/qstrdefs.generated.h: $(SDKCONFIG_H) $(BOARD_DIR)/mpconfigboard.h
+ $(HEADER_BUILD)/mpversion.h: $(SDKCONFIG_H)
```

nevercast, 2019-09-11:
Confirmed, removing that line, build fails with `sdkconfig.h: No such file or directory`
Putting the line back is build success.

dpgeorge, 2019-09-11:
See #5092 for a fix