micropython/micropython #5098: UART timeout and timeout_char is not working in ESP32

WanderGit, 2019-09-12:
Hi,

I noticed that UART timeout_char is not working. Regardless of the value of this parameter, UART always waits for around 8x the time of a character to identify that communication has ended.
With the help of a forum contributor, we were able to locate and fix the problem.
In the machine_uart.c we add the following command to line 208:

    uart_set_rx_timeout(self->uart_num, self->timeout_char);

==================================================
Now that code snippet looked like this:

    // set timeout_char
    // make sure it is at least as long as a whole character (13 bits to be safe)
    self->timeout_char = args[ARG_timeout_char].u_int;
    uint32_t min_timeout_char = 13000 / baudrate + 1;
    if (self->timeout_char < min_timeout_char) {
        self->timeout_char = min_timeout_char;
    }
    uart_set_rx_timeout(self->uart_num, self->timeout_char);

==================================================

I don't know if timeouts have been disabled for a specific reason, but I would like to report that this minor fix resolves the issue, at least for timeout_char.

dpgeorge, 2019-09-12:
Thanks for the report and fix.

`timeout_char` (like `timeout`) is measured in milliseconds, but the argument to `uart_set_rx_timeout` is in symbols/characters.  So there'll need to be a conversion, from ms to chars.

robert-hh, 2019-09-12:
@WanderGit Thank you for taking my post content and putting it into a PR. I forgot to ask if someone could do that, because I do not like the trouble with git and PRs. Sooner or later my repos get out of sync.

WanderGit, 2019-09-15:
Hi @robert-hh 
This is my first open issue ever :smile:

I took so long trying to understand what was going on and when I asked for help on the forum the help came so fast!
So I wanted to warn others about the problem and how to fix it!
Thank you very much for your help!

baeverell, 2021-05-10:
Hello, is this fix implemented in the repository? If not, can you provide some steps for how to create a workaround? I am having some trouble receiving UART messages on ESP32. Thanks in advance.

robert-hh, 2021-05-10:
I can't see a place where the timeout_char value is actually used. The call mentioned above to uart_set_rx_timeout() is not implemented. The only thing implemented is timeout.
You can use python code to wait a while for incoming characters.

baeverell, 2021-05-10:
could you provide an example please?
Is it simple as adding time.sleep(1) in between uart reads?

robert-hh, 2021-05-10:
How do you want to receive messages? The timeout setting should work.