micropython/micropython #5109: Performance of ESP32 daily builds compared to release 1.11

ramprao, 2019-09-15:
On an ESP32, I am getting a considerable slowdown when I compare the performance of the current 1.11 daily build (v1.11-312-g22099ab88) with the v1.11 release build from 20190529. The tests I am using are the ledN tests from Damien's talk [Writing fast and efficient MicroPython at Pycon AU 2018](https://www.youtube.com/watch?v=hHec4qL00x0). 

Here is the comparison. Not sure what is the cause. Part of the motivation was to try and see if I could build the recent native support PR for the ESP32.  

### With standard release firmware
MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32

```
>>> import led1
7.007 sec, 35.035 usec/blink :    28.54 kblinks/sec

>>> import led2
6.434 sec, 32.171 usec/blink :    31.08 kblinks/sec

>>> import led3
0.917 sec,  4.586 usec/blink :   218.03 kblinks/sec

>>> import led4
0.755 sec,  3.774 usec/blink :   264.96 kblinks/sec
```

### With daily build 
MicroPython v1.11-312-g22099ab88 on 2019-09-15; ESP32 module with ESP32

```
>>> import led1
11.800 sec, 58.999 usec/blink :    16.95 kblinks/sec

>>> import led2
11.212 sec, 56.062 usec/blink :    17.84 kblinks/sec

>>> import led3
6.444 sec, 32.221 usec/blink :    31.04 kblinks/sec

>>> import led4
1.446 sec,  7.230 usec/blink :   138.31 kblinks/sec
```

### With daily build
MicroPython v1.11-312-g22099ab88 on 2019-09-15; ESP32 module with ESP32
But do not define r = range(n). The speed seems to be somewhat higher. 

```
>>> import led3a
3.480 sec, 17.401 usec/blink :    57.47 kblinks/sec

>>> import led4a
1.075 sec,  5.377 usec/blink :   185.99 kblinks/sec
```

andrewleech, 2019-09-15:
The esp32 IDF version has changed a couple of times since the release, fair chance it might be something there making the bulk of the difference.

That commit looks like it includes the very recent update to ESP IDF v3.3
https://github.com/micropython/micropython/commit/5641aa55dddbdf312a1e8dbb8c3936115683840a#diff-77cbec69bd2e6d723afa755b9d99031b
It would be interesting to compare your results to the commit just before that.

ramprao, 2019-09-16:
Indeed it appears that there is a significant improvement for all tests. The led3 and led4 tests are back to values same as 1.11 release while the led1 and led2 are still a little lower in speed. I used [c69f58e6b97b954d06734797f06feeab4e510855](https://github.com/micropython/micropython/commit/c69f58e6b97b954d06734797f06feeab4e510855).
```
>>> import led1
9.368 sec, 46.839 usec/blink :    21.35 kblinks/sec
>>> import led2
8.790 sec, 43.949 usec/blink :    22.75 kblinks/sec
>>> import led3
0.918 sec,  4.591 usec/blink :   217.82 kblinks/sec
>>> import led4
0.755 sec,  3.775 usec/blink :   264.88 kblinks/sec
```

dpgeorge, 2019-09-16:
Note that performance of the esp32 varies a lot, see a previous comment about this here: https://github.com/micropython/micropython/issues/5040#issuecomment-524334492

ramprao, 2019-09-16:
To add some more information for the c69f58e6b firmware, I get slightly different results for the led1 and led2 tests between alternate reboots. Here are the values. 

For led1:
```
MPY: soft reboot
>>> import led1
11.719 sec, 58.594 usec/blink :    17.07 kblinks/sec
MPY: soft reboot
>>> import led1
9.366 sec, 46.828 usec/blink :    21.35 kblinks/sec
MPY: soft reboot
>>> import led1
11.719 sec, 58.593 usec/blink :    17.07 kblinks/sec
MPY: soft reboot
>>> import led1
9.366 sec, 46.828 usec/blink :    21.35 kblinks/sec
```
For led2:
```
MPY: soft reboot
>>> import led2
11.142 sec, 55.711 usec/blink :    17.95 kblinks/sec
MPY: soft reboot
>>> import led2
8.790 sec, 43.949 usec/blink :    22.75 kblinks/sec
MPY: soft reboot
>>> import led2
11.142 sec, 55.712 usec/blink :    17.95 kblinks/sec
MPY: soft reboot
>>> import led2
8.790 sec, 43.949 usec/blink :    22.75 kblinks/sec
``` 
If I do a machine.reset() between tests, I consistently get 17.07 kblinks/sec for led1 test and 17.95 kblinks/sec for led2 test. The led3 and led4 tests are always the same.

dpgeorge, 2021-10-15:
This should be fixed by 549448e8bbc8ce0b6b5fc51c0660acdaff18c3d6