micropython/micropython #5110: Availability of sys.exc_info() in current builds

amotl, 2019-09-16:
Dear @pfalcon and the MicroPython community,

we recently received a pyboard-D and upgraded the firmware to the latest release so we are now running
```
MicroPython v1.11-312-g22099ab88 on 2019-09-15; PYBD-SF2W with STM32F722IEK
```

We are now aiming at porting our datalogger program [1] from Pycom MicroPython to the latest Genuine MicroPython coming from your pen.

However, we are having hard times figuring out why MicroPython croaks on this code
```python
>>> import sys
>>> sys.exc_info()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'exc_info'
```

Does MicroPython just doesn't support `sys.exc_info()` at all or did we miss something differently?

Thanks in advance for taking the time to look at this issue.

With kind regards,
Andreas.

[1] https://github.com/hiveeyes/hiveeyes-micropython-firmware

jimmo, 2019-09-16:
`sys.exec_info` is supported by MicroPython but disabled by default for most ports. The pycom fork of MicroPython enables it for their ESP32 builds.

If you are prepared to build the firmware yourself (see instructions in ports/stm32/README.md) then you can define your own board and add `#define MICROPY_PY_SYS_EXC_INFO (1)`. (Or just add this to the PYBD_SF2 config). See the SF3 and SF6 directories for examples on how to extend from another board.

amotl, 2019-09-16:
Dear @jimmo,

we've already found the `MICROPY_PY_SYS_EXC_INFO` constant while grepping through the code but as relative beginners, we haven't yet built a firmware for ourselves.

We will either go down that route or adapt the Python code to a different error handling pattern by propagating the exception object into `log.exc()`.

Thank you very much for your clear answer about that topic and for outlining the process of building a custom firmware version.

With kind regards,
Andreas.

amotl, 2019-09-16:
> Adapt the Python code to a different error handling pattern by propagating the exception object into `log.exc()`.

That would probably be the better option when aiming at making a codebase work out of the box for a larger audience of people, right?

dpgeorge, 2019-09-16:
> That would probably be the better option when aiming at making a MicroPython codebase work out of the box for a larger audience of people, right?

Yes, definitely, that's a more robust way to get at the current exception.

The reason not to enable `SYS_EXC_INFO` is because `sys.exc_info()` is not 100% compatible with the CPython version, and it doesn't work with the native code emitter.  Not enabling it forces users to find another way to solve the problem of getting the exception and means we can just leave it disabled.

amotl, 2019-09-17:
Dear @dpgeorge and @jimmo,

thanks for sharing some more background knowledge about this specific detail. We see your points and will be happy to adapt our codebase accordingly.

It's a bit sad that we started with Pycom MicroPython here which enables this option and now we have to move back, but in fact it's yet another _c'est la vie_ thing ;].

Thanks again for the valuable information and keep up the spirit.

With kind regards,
Andreas.

closed 2019-09-17 by amotl