micropython/micropython #5113: esp32 ppp lib support   sim7600   4g?

coffice12, 2019-09-17:


MrSurly, 2019-09-17:
PPP is PPP.  Your cell module must support PPP.  In my experience, getting your "[chat](https://linux.die.net/man/8/chat)" script correct is the hardest part.

coffice12, 2019-09-17:
I want to use esp32 to access sim7600 to realize 4G ap. What's a good idea?

MrSurly, 2019-09-17:
```
class ChatMsg:
    def __init__(self, waitfor, msg, timeout = 5):
        self.waitfor = bytes(waitfor, 'utf-8')
        self.msg = bytes(msg, 'utf-8')
        self.timeout = timeout

class Chat:
    class cont(Exception):pass

    def __init__(self, uart, cmds, debug = False):
        self.uart = uart
        self.cmds = cmds
        self._debug = debug

    @property
    def debug(self):
        return self._debug

    @debug.setter
    def debug(self, value):
        self._debug = value

    def run(self):
        for cmd in self.cmds:
            if cmd.waitfor:
                if self._debug:
                    print('await(%d) "%s"' % (cmd.timeout, str(cmd.waitfor)), end='')
                acc = bytearray()
                t = time.time() + cmd.timeout
                try:
                    while(time.time() < t):
                        acc += (self.uart.read() or b'')
                        if cmd.waitfor in acc:
                            raise Chat.cont
                    return False
                except Chat.cont:
                    if self._debug:
                        print('-->', cmd.msg)
                    self.uart.write(cmd.msg)
                    continue
            else: # nothing to waitfor
                if self._debug:
                    print('sleep(%d) -->' % cmd.timeout, cmd.msg)
                time.sleep(cmd.timeout)
                self.uart.write(cmd.msg)

        return True
```

Here's a py Chat class I wrote that is similar to the *nix [chat](https://linux.die.net/man/8/chat) command.  You pass it a UART, and an array of `ChatMsg`.  

`ChatMsg` initializer takes  3 parameters `waitfor`, `msg`, and `timeout`.

`waitfor` is a literal string to be received over the wire before sending `msg`.  It waits a maximum of `timeout` seconds.  `waitfor` can be empty, which means "dont wait for anything, just send `msg`".  In this case (`waitfor` being empty), then the timeout is how long to pause _after_ sending `msg`.

MrSurly, 2019-09-17:
> I want to use esp32 to access sim7600 to realize 4G ap. What's a good idea?

1) Ensure you can physically connect this module to your ESP32 module.  You should have a working UART connection (i.e. you can send AT commands and get responses)
2) Discover the correct sequence of AT commands that command the module to enter PPP mode.  You're likely going to have to make sure of signal strength and network registration as a prerequisite to sending these commands.
3) Use the knowledge from (2) to write code that automates this process to make a reliable connection.

After PPP is setup and connected, then all the network stuff "just works."

Getting PPP working in any environment (be it ESP32 or Windows, Linux, MAC, or anywhere else) isn't particularly _difficult_, but there is a learning curve.

coffice12, 2019-09-17:
Thank you very much for trying.

closed 2019-09-17 by coffice12

MrSurly, 2019-09-17:
I forgot to mention, you'll need information from your SIM card provider about the correct PDP context , and use the AT+CGDCONT command to set up that context.