micropython/micropython #5114: ESP32: SPIRAM support disappeared

robert-hh, 2019-09-17:
Something changed with the latest commits for ESP32. The support for SPIRAM disappeared. I can build for GENERIC_SPIRAM, but gc.mem_free() does not show any more the external RAM. It works with b45f9de8096f37a5f588f833e6b6a00b59e67e3f

jimmo, 2019-09-18:
Oh sorry, the variable is renamed in 4.0 so I added both definitions to the sdkconfig.spiram and changed the code to use the new one, but it appears that only the one that the IDF recognises ends up in sdkconfig.h.

The fix is

```
diff --git a/ports/esp32/main.c b/ports/esp32/main.c
index 4d25cf0b3..7106e0bf5 100644
--- a/ports/esp32/main.c
+++ b/ports/esp32/main.c
@@ -74,7 +74,8 @@ void mp_task(void *pvParameter) {
     #endif
     uart_init();
 
-    #if CONFIG_ESP32_SPIRAM_SUPPORT
+    // TODO: CONFIG_SPIRAM_SUPPORT is for 3.3 compatibility, remove after move to 4.0.
+    #if CONFIG_ESP32_SPIRAM_SUPPORT || CONFIG_SPIRAM_SUPPORT
     // Try to use the entire external SPIRAM directly for the heap
     size_t mp_task_heap_size;
     void *mp_task_heap = (void*)0x3f800000;
```

I will send a PR now. Sorry for the inconvenience!

dpgeorge, 2019-09-18:
Fixed by 62d78e231ceb41eecdc3180cae6f07d790cd33d8

closed 2019-09-18 by dpgeorge