micropython/micropython #5119: ESP32: "phy: phy_version" log message in raw mode

nickzoic, 2019-09-18:
Log messages are supposed to all be suppressed in raw mode (See PR #3817)
but this "phy: phy_version" message sneaks through.  The interjected log message upsets all the usual REPL-bashing tools like ampy and mpy-utils.

This is on the official 1.11 build from https://micropython.org/downloads#esp32 . Brought to my attention by Seon @UnexpectedMaker and Jim @jimmo 

To replicate, go into raw REPL with ctrl-A:

```MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
Type "help()" for more information.
>>>
raw REPL; CTRL-B to exit
>```

At raw repl, type (invisibly):
```
import network
network.WLAN().active(1)
```
... then hit Ctrl-D:

```OKI (9836120) phy: phy_version: 4007, 9c6b43b, Jan 11 2019, 16:45:07, 0, 0
>
```

nickzoic, 2019-09-18:
Related discussion on ESP-IDF repo at https://github.com/espressif/esp-idf/issues/1855 was closed, but without fixing the root cause

dpgeorge, 2019-09-26:
Hmm, so one option would be to just disable logging, using `ESP_LOG_NONE`.  But that's probably a last resort since logging is useful.

I wonder if we can define `LOG_LOCAL_LEVEL` to be a dynamic value so the level can be changed at runtime?  Eg:
```
#define LOG_LOCAL_LEVEL (log_local_level_variable)
int log_local_level_variable;
```
That will probably make the firmware really big because the levels can't be optimised out by the compiler.

Failing that we would need to provide our own implementation of the `lib_printf.c` file.

closed 2026-03-19 by agatti