micropython/micropython #5121: import * allowed at function level

petrkr, 2019-09-18:
Hello,
micropython allows import * from modules in function level. Is this desired behavior?

micropython behavior: 

```pycon
>>> def test():
...   from os import *
... 
>>> 
>>> 
__class__       __name__        gc              uos
bdev            test
>>> 
>>> 
>>> test()
>>> 
>>> 
__class__       __name__        remove          VfsFat
chdir           dupterm         dupterm_notify  gc
getcwd          ilistdir        listdir         mkdir
mount           rename          rmdir           stat
statvfs         umount          uname           uos
urandom         bdev            test
>>> 
```

expected behavior (CPython):

```pycon
$ python3.4
Python 3.4.6 (default, Mar 26 2017, 13:41:08) 
[GCC 4.9.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> def test():
...   from os import *
... 
  File "<stdin>", line 1
SyntaxError: import * only allowed at module level
```

stinos, 2019-09-19:
There's no explicit checking for this in the code, likely just because MicroPython is optimized to be small/efficient.

robert-hh, 2019-09-19:
That's right. The Python reference does not allow it. But, on the other hand, this is not a deviation that hurts, since using "from module import *" works at module level as it should.

dpgeorge, 2019-09-19:
I didn't know that this was disallowed inside a function/class definition... I guess the technical reason is that doing an `import *` doesn't know the names (at compile time) that will be imported so cannot store them to the local variables within a function.  That's why in the example above they leak into the global scope.

This would be relatively easy to "fix" to just provide an error in the compiler, but I don't see that it's really important to disallow since it should anyway never be used (it won't be compatible with CPython) and doesn't crash or put MicroPython into an undefined state.

It could be added as a difference in `tests/cpydiff`.

encukou, 2019-09-20:
I've seen MicroPython-only code that relies on this behavior. It was quite "interesting" to debug :)

> I don't see that it's really important to disallow since it should anyway never be used (it won't be compatible with CPython) and doesn't crash or put MicroPython into an undefined state.

The same could be said for `return`. If I comment out these lines in `compile.py`:
```
        //compile_syntax_error(comp, (mp_parse_node_t)pns, "'return' outside function");
        //return;
```
then modules & classes can return:
```console
$ cat demo.py
print('Will return...')
return 42
print('Hello')

$ ./micropython demo.py 
Will return...
$
```

I've started the "fix" in #5130, but being new to MicroPython, I'm stuck figuring out how to test this.

peterhinch, 2019-09-20:
Your script:
```python
print('Will return...')
return 42
print('Hello')
```
produces the same result on import in MicroPython as in CPython:
```
SyntaxError: 'return' outside function
```
MicroPython's existing behaviour is correct.

petrkr, 2019-09-20:
It was encukou's example for situation. For already deleted comment..

Someone said that quoted message and encukou demostrated it on different situation which is checked, so why this * not.

dpgeorge, 2019-10-04:
Fixed by 25a9bccdee2fe830046c1c1a0220ca7706597202

closed 2019-10-04 by dpgeorge