micropython/micropython #5123: bytes.find() differs from CPython

peter-conalgo, 2019-09-19:
In Cpython (at least) I can find by integer using `bytes.find()`. Micropython doesn't allow this and only accepts`bytes` as an argument to find.

Micropython:

```
>>> s = b'ABC'
>>> s.find(b'B')
1
>>> s.find(66)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'int' object to str implicitly
>>> 
```

Desktop python:

```
>>> s = b'ABC'
>>> s.find(b'B')
1
>>> s.find(66)
1
```

Here is the check that's failing, but additional code would be needed to convert int (0..255) into byte.

<https://github.com/micropython/micropython/blob/master/py/objstr.c#L693>

This bug bit me deep in some QR encoding logic.

dpgeorge, 2019-09-19:
Thanks for the report.  It's clear that it is a difference to CPython.

> This bug bit me deep in some QR encoding logic.

Can you explain the bug, was it that it just raised an exception that it couldn't convert int to str (so easy to find and workaround), or something else?

pmp-p, 2019-09-20:
i think he meant bytes.find() is behaving as str.find() as in there's only str_finder() and no bytes_finder() where a single int should be implicitly converted to a bytes length 1

peterhinch, 2019-09-20:
A bytes instance may be viewed as a string or a list of small integers depending on context. It therefore seems reasonable that you should be able to write
```python
data = b'\x01\x02\x03'
data.find(0x02)
```
Which works in CPython but not MicroPython.

peter-conalgo, 2019-09-20:
> Can you explain the bug, was it that it just raised an exception that it couldn't convert int to str (so easy to find and workaround), or something else?

Yes, raises _TypeError_ like example above. Issue happens <https://github.com/JASchilz/uQR/blob/master/uQR.py#L902-L905> but I've already worked around it in my version of that code.

dpgeorge, 2019-09-23:
> Yes, raises _TypeError_ like example above.

Ok, so at least the exception stopped the code and was easy to find, rather than silently doing something different/unexpected.

Eventually it might be worth fixing this, depending on how complicated the fix is and how useful this feature is.