micropython/micropython #5125: [ ESP32 ] How to change the Maximum simultaneously WiFi connection on the softAP?

beyonlo, 2019-09-19:
Hello.

I tested and the Maximum is 4, right?

**How I tested:**
1. I configured my ESP32 to AP mode.
2. I want to have 10 other ESP32 STA connected simultaneously in the ESP32 AP.
3. I have error in the ESP AP when the ESP STA number 5 tried to connect to AP, with error: **wifi: max connection, deauth**

**Follow the sequence error in the AP:**

MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
Type "help()" for more information.
>>> import network
>>> ap = network.WLAN(network.AP_IF)
I (1342589) wifi: wifi driver task: 3ffc4830, prio:23, stack:3584, core=0
I (1342589) wifi: wifi firmware version: 38e2484
I (1342589) wifi: config NVS flash: enabled
I (1342599) wifi: config nano formating: disabled
I (1342599) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1342609) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1342639) wifi: Init dynamic tx buffer num: 32
I (1342639) wifi: Init data frame dynamic rx buffer num: 32
I (1342639) wifi: Init management frame dynamic rx buffer num: 32
I (1342639) wifi: Init static rx buffer size: 1600
I (1342649) wifi: Init static rx buffer num: 10
I (1342649) wifi: Init dynamic rx buffer num: 32
>>> ap.config(essid='esp', password='esp')
>>> ap.active(True)
I (1353999) phy: phy_version: 4007, 9c6b43b, Jan 11 2019, 16:45:07, 0, 0
I (1354009) wifi: mode : softAP (24:6f:28:de:40:85)
I (1354009) wifi: Init max length of beacon: 752/752
I (1354009) wifi: Init max length of beacon: 752/752
True
>>> I (1354019) network: event 13
print(ap.ifconfig())
('192.168.4.1', '255.255.255.0', '192.168.4.1', '0.0.0.0')
>>> I (1540579) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (1540579) wifi: station: 0c:cb:85:3e:15:c1 join, AID=1, bgn, 20
I (1540579) network: event 15
dhcps: send_offer>>udp_sendto result 0
I (1540959) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (1540959) network: event 17
I (1551259) wifi: station: 0c:cb:85:3e:15:c1 leave, AID = 1, bss_flags is 134259
I (1551259) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (1551259) network: event 16
I (1556339) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (1556339) wifi: station: 0c:cb:85:3e:15:c1 join, AID=1, bgn, 20
I (1556339) network: event 15
dhcps: send_offer>>udp_sendto result 0
I (1556949) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.2
I (1556949) network: event 17
I (1638929) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (1638929) wifi: station: b4:ef:39:84:80:50 join, AID=2, bgn, 20
I (1638929) network: event 15
dhcps: send_offer>>udp_sendto result 0
I (1639649) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.3
I (1639659) network: event 17
I (1677649) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (1677649) wifi: station: 08:c5:e1:c4:c5:5a join, AID=3, bgn, 20
I (1677649) network: event 15
dhcps: send_offer>>udp_sendto result 0
I (1678469) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.4
I (1678469) network: event 17
I (1801719) wifi: new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (1801719) wifi: station: 70:fd:46:ab:86:3c join, AID=4, bgn, 20
I (1801729) network: event 15
dhcps: send_offer>>udp_sendto result 0
I (1802339) tcpip_adapter: softAP assign IP to station,IP is: 192.168.4.5
I (1802339) network: event 17
I (1923569) wifi: max connection, deauth!
I (1923879) wifi: max connection, deauth!
I (1924189) wifi: max connection, deauth!
I (1924499) wifi: max connection, deauth!
I (1924819) wifi: max connection, deauth!
I (1925129) wifi: max connection, deauth!
I (1925439) wifi: max connection, deauth!
I (1927019) wifi: max connection, deauth!
I (1927329) wifi: max connection, deauth!
I (1927649) wifi: max connection, deauth!
I (1927959) wifi: max connection, deauth!
I (1928269) wifi: max connection, deauth!
I (1928589) wifi: max connection, deauth!

beyonlo, 2019-09-20:
Hello.

Here https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv416wifi_ap_config_t say about: 

>uint8_t max_connection
>Max number of stations allowed to connect in, default 4, max 4

So, is not possible use ESP32 AP with more than 4 clients connected?

Are there a way to change that? I would like 10.

Thank you.

MrSurly, 2019-09-20:
> Are there a way to change that? I would like 10.

This is a limitation with the IDF; you should submit this issue to Espressif.

dpgeorge, 2019-09-23:
It looks like it could be up to 10, according to the constant `ESP_WIFI_MAX_CONN_NUM`

beyonlo, 2019-09-23:
> It looks like it could be up to 10, according to the constant `ESP_WIFI_MAX_CONN_NUM`

@dpgeorge 

How? Because **uint8_t max_connection** is **Max number of stations allowed to connect in, default 4, max 4** following https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv416wifi_ap_config_t

So, if max is 4 in the IDF conf, how **ESP_WIFI_MAX_CONN_NUM** can be more than that IDF config?

ShenTengTu, 2019-11-12:
> uint8_t max_connection
> Max number of stations allowed to connect in, default 4, max 4

In [esp-idf/components/wpa_supplicant/src/ap/ap_config.h](https://github.com/espressif/esp-idf/blob/a45e9985344575a80acb1dc9c12e7bec4d8af401/components/wpa_supplicant/src/ap/ap_config.h#L15), the constant `MAX_STA_COUNT` is 4. 

This constant is used in [`void hostapd_config_defaults_bss(struct hostapd_bss_config *bss)`](https://github.com/espressif/esp-idf/components/wpa_supplicant/src/ap/ap_config.c#L39)

-------

According to the description of field `max_connection` in ["AP Basic Configuration"](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#ap-basic-configuration) : 
> Currently, ESP32 Wi-Fi supports up to 10 Wi-Fi connections. If max_connection > 10, AP defaults the value to 10.

It also described that API [`esp_wifi_set_config`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_set_config16wifi_interface_tP13wifi_config_t) can be used to configure the AP. 

Currently, `WLAN.config()` in Micropython has no "max_connection`"parameter. Is it possible to implement connection number limit feature?

ipapagi, 2019-12-29:
@ShenTengTu any news about `max_connection`?

dpgeorge, 2020-01-22:
Implemented by a55c17dc69ac43cd41e7efe603e20298a5513b06

closed 2020-01-22 by dpgeorge

FPOOOOOOO, 2022-07-18:
When and How can the connection number up to 16

dpgeorge, 2022-07-19:
> When and How can the connection number up to 16

The maximum is 10, you cannot configure more than that.

FPOOOOOOO, 2022-07-19:
> > When and How can the connection number up to 16
> 
> The maximum is 10, you cannot configure more than that.

Is the maximum 10 limit by hardware or software. If it's the software I believe there is a way to enlarge it.

dpgeorge, 2022-07-19:
> Is the maximum 10 limit by hardware or software.

I do not know.  The limit of 10 is set by the ESP IDF.  You will need to ask there.

commit a55c17dc (2020-01-22): esp32/modnetwork: Add max_clients kw-arg to WLAN.config for AP setting.

This allows the user to configure the maximum number of clients that are
connected to the access point.  Resolves #5125.