micropython/micropython #5128: esp32: "Chip revision is higher than the one configured..."

nevercast, 2019-09-19:
Using an ESP32-PICO chip, building against this version of MicroPython https://github.com/micropython/micropython/commit/d96391aca34eb4d516945b34f0bbebf807c41299

I'm seeing a new warning in the debug output. 
`W (155) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.`

[image]

How can I upgrade my targetted chip revision?

jimmo, 2019-09-19:
See the tinypico board config (sdkconfig.board).

nevercast, 2019-09-20:
I checked the TinyPICO config;

I created an `sdkconfig.esp32rev1` file, and added the line
```
CONFIG_ESP32_REV_MIN_1=y
```

Then in my board config `mpconfigboard.mk` file I added the line
```
SDKCONFIG += $(BOARD_DIR)/../sdkconfig.esp32rev1
```

I performed a `make clean` and `make`, followed by flashing the image. The warning is still present.

nevercast, 2019-09-20:
Also no luck with 
```
CONFIG_ESP32_REV_MIN_1=y
CONFIG_ESP32_REV_MIN=1
```

nevercast, 2019-09-20:
Quick var dump gets me no closer
```
$SDKCONFIG is [/home/josh/Desktop/esp32/upy/lora/micropython/ports/esp32/boards/sdkconfig.base /home/josh/Desktop/esp32/upy/lora/micropython/ports/esp32/boards/sdkconfig.spiram /home/josh/Desktop/esp32/upy/lora/src/boards/hub/../sdkconfig.ipcextended /home/josh/Desktop/esp32/upy/lora/src/boards/hub/../sdkconfig.esp32rev1]
$CONFIG_ESP32_REV_MIN is [1]
$CONFIG_ESP32_REV_MIN_0 is []
$CONFIG_ESP32_REV_MIN_1 is [y]
```

nevercast, 2019-09-20:
Seems it was insufficient to leave `CONFIG_ESP32_REV_MIN_0` undefined, I had to explicitly set it to `n`

```
CONFIG_ESP32_REV_MIN_0=n
CONFIG_ESP32_REV_MIN_1=y
```

It is sufficient to omit `CONFIG_ESP32_REV_MIN=1`, this is handled by KConfig.

nevercast, 2019-09-20:
Before I close this issue as resolved, @UnexpectedMaker are you seeing this message on your TinyPICOs? As the one line `CONFIG_ESP32_REV_MIN_1` was insufficient for my PICO board.

UnexpectedMaker, 2019-09-20:
@nevercast - I've never seen that before, no. All my IC's are rev1, I check them in the test jig to be 100% sure, but that log has never appeared for me.

ramprao, 2019-09-21:
I'm getting this message on an older NodeMCU ESP32  DevKitV1 board with recent firmware. This appears to be related to the switch to ESP IDF 3.3 --  [5641aa5](https://github.com/micropython/micropython/commit/5641aa55dddbdf312a1e8dbb8c3936115683840a). The build just before --  [c69f58e](https://github.com/micropython/micropython/commit/c69f58e6b97b954d06734797f06feeab4e510855) -- appears to not show this.

nevercast, 2019-09-24:
Sounds like we might want to add the lines 
```
CONFIG_ESP32_REV_MIN_0=n
CONFIG_ESP32_REV_MIN_1=y
```
In to the GENERIC board for the ESP32 port, since most ESP32s out there are rev1 or greater. But let's wait to see what Espressif do here first https://github.com/espressif/esp-idf/issues/4000

At time of writing they are pending some changes to go public.

jeremyherbert, 2019-10-09:
Yes, this was due to a change in ESP-IDF (I had some other projects that started showing up this error).

The only ESP32 that is rev0 that I have was from before the ESP32 was FCC certified (which was a long time ago). And actually the reason I found out it was rev0 is because micropython didn't work on it ;)

nevercast, 2019-10-09:
Hi @igrr, now that https://github.com/espressif/esp-idf/commit/dd248ffc32fd248031cefb505629e034dd329bb9 has landed in esp-idf. Do we no longer need 

```
CONFIG_ESP32_REV_MIN_0=n
CONFIG_ESP32_REV_MIN_1=y
```

In our configuration? Just want to confirm the expected configuration with regards to the warning message reported in https://github.com/espressif/esp-idf/issues/4000

Cheers, Josh.

igrr, 2019-10-09:
@nevercast The configuration depends on the chip revisions you want to have supported in the app. 

Setting the "minimal supported chip revision" to 1 will result in an application binary which will not run on revision 0 chips. It will still run on later (revision >=1) chips.

If you want to have a single binary which can run on all revisions of the chip, set this option to 0.

nevercast, 2019-10-09:
Thanks @igrr,

1) will we still get the warning, confusing users.
2) will later revisions be impacted if we compile for old revision compatibility 
3) will older revision compatibility increase code size?

igrr, 2019-10-09:
1) It looks like the message was changed to "Info" level in the commit you have referenced. If you think the message can be worded in a less confusing way, please propose such wording, we will update it.
2) By default we write IDF code in a way to support all chip revisions. So enabling compatibility with rev. 0 is always the safest option. Code compiled for rev. 0 will continue to work on the future chip revisions. Any cases of incompatibility between new and old revisions will be listed when that new revision is announced (no such cases known so far).
3) Compiling with a higher minimum supported revision will in general decrease the code size. However there won't be any significant difference between rev 0 and rev 1. For the later revisions, you may indeed notice some code size reduction.

dpgeorge, 2019-10-10:
Since we now have board configs for esp32 I think it's reasonable that 1) the default GENERIC boards should support all chip revisions; 2) other boards can configure these options as they see fit (eg TINYPICO).

The GENERIC boards are intended to be easy to use for most cases, so should support rev0.  Even if it gives a warning... we just have to educate users, or get them to use a board that is better tailored to their specific board.

nevercast, 2019-10-10:
Thanks igrr for the helpful reply.

I agree @dpgeorge, I do think we should support rev0 on the GENERIC board. A small problem that crops up is that it seems the fix for this warning isn't backported.

```
git tag --contains dd248ffc
<EMPTY>
```

```
git branch --contains dd248ffc
* master
```

This means that all our esp-idf 3.x builds (our default) will still have the warning.

@igrr Do you have a tag or v3.x branch that you backport fixes to? Could https://github.com/espressif/esp-idf/issues/4000 also be backported?

RobertAtLegalitySoftware, 2020-05-26:
Hello all.  I'm a brand-newbie to MicroPython, and the ESP32.  So I apologize if I'm swimming in the deep end here, but I am having this same issue.  I found several threads on this topic but this one seems closest to my scenario.

My ESP32 is from [Amazon](https://www.amazon.com/gp/product/B07Q576VWZ) and has the WROOM32 metal can.  I'm using Thonny and after fighting through roadblocks for 2 days I finally have a blink sketch working.  

I believe I understand the version issue you are describing.  My chip is newer than the lowest version supported, but I can't figure out where to find the settings file I need to change to add the fix you describe here.  Can someone please help me out?
Thanks! - Robert

jeremyherbert, 2020-05-26:
Can you post a photo? You may have one of the new WROOM-E devices

robert-hh, 2020-05-26:
As mentioned above, this is just an informal message. It does not affect the device operation. The firmware image for the Generic ESP32 device (e.g. ones with the WROOM module) is build in a way, that it works with any ESP32 chips from Revision 0 up.

RobertAtLegalitySoftware, 2020-05-26:
> Can you post a photo? You may have one of the new WROOM-E devices
Sure.  Here you go.  Thanks!
[image]

RobertAtLegalitySoftware, 2020-05-26:
> As mentioned above, this is just an informal message. It does not affect the device operation. The firmware image for the Generic ESP32 device (e.g. ones with the WROOM module) is build in a way, that it works with any ESP32 chips from Revision 0 up.
I understand it is just a warning but I really want to know what file should be modified to specify the version config value.  I'm just trying to learn 'all the things' and mark this puzzle off my list.  Thanks Roberthh.

robert-hh, 2020-05-26:
> I understand it is just a warning but I really want to know what file should be modified to specify the version config value. 

You have to build your own firmware image and include the lines:
```
CONFIG_ESP32_REV_MIN_0=n
CONFIG_ESP32_REV_MIN_1=y
```
into ports/esp32/boards/sdkconfig.base. You can also put that into a separate file and include that into
ports/esp32/boards/GENERIC/mpconfigboard.mk

RobertAtLegalitySoftware, 2020-05-26:
> You have to build your own firmware image...

Ah!  That makes sense.  I was thinking there was a config file somewhere I couldn't find.  It will be a while before I'm deep enough to tackle compiling custom firmware :) but I really appreciate you taking time to clear up my confusion.  Every little lesson helps...

Orel-A, 2020-07-08:
To those who came to this post like me and are new to MicroPython; no there isn't any menu config that you need to toggle in python.
This is irrelevant for newbies as this requires to recompile the entire project of MicroPython so just ignore the suggestion

burkulesomesh43, 2020-07-27:
Hello everyone,

Is there any issue if we cannot set current chip revision in menuconfig?

jimmo, 2020-07-28:
> To those who came to this post like me and are new to MicroPython; no there isn't any menu config that you need to toggle in python.
> This is irrelevant for newbies as this requires to recompile the entire project of MicroPython so just ignore the suggestion

That's right -- this is just a warning message, there is nothing wrong.

MicroPython doesn't use menuconfig, however the sdkconfig files that are being described are what menuconfig would normally generate, so modifying these files as described above is equivalent to changing this setting in menuconfig.

> Is there any issue if we cannot set current chip revision in menuconfig?

@burkulesomesh43 not quite sure what you're asking, but I don't think so. Unfortunately our choice is:

a) provide a different prebuilt generic firmware image for every chip revision (confusing for users)
b) some users see this message (which can be safely ignored)

Which is why we went with option (b).

And for the users where this really matters (i.e. people making ESP32 boards who want to provide their own MicroPython firmware) they have the option like TINYPICO to define a custom configuration for their board.

closed 2020-07-28 by jimmo

burkulesomesh43, 2020-07-28:
Hi @jimmo ,

I am asking about it because I am facing an issue with esp32 which may hardfault or stucked in brown out.
As I checked esp32 errata, And it is documented that..

**3.4. The Brown-out Reset (BOR) function does not work. The system fails to boot up after BOR.
Workarounds:
There is no workaround for this issue.
Fixes:
This issue is fixed in silicon revision 1.** 

As in my firmware, the default config for chip revision was **Rev 0** and my esp32 module have chip revision 1.
So is there any issue if cannot change it?
silicon revision and chip revision is same?