micropython/micropython #5136: Reconfigure REPL key bindings for emacs users

nealmcb, 2019-09-23:
When I use the micropython REPL, my fingers (which are trained in 3 decades of Emacs use) habitually hit `Ctrl-a` all the time, expecting to navigate to the beginning of the line. But this puts micropython in RAW mode instead and leads to lots of wasted time.

Is there any way to change the keybindings used in the Micropython REPL?

It seems this may be related to decisions around GNU readline (which used to be used here under some circumstances?) and supported emacs keybindings?

dpgeorge, 2019-09-23:
> hit `Ctrl-a` all the time, expecting to navigate to the beginning of the line. But this puts micropython in RAW mode instead

It should only enter raw mode if the line is empty.  In all other cases it should go to the start of the line.

> Is there any way to change the keybindings used in the Micropython REPL?

At this point you can only do this by editing the source code in the relevant places (and there are lots of places, so it's not that easy to do).

> It seems this may be related to decisions around GNU readline

GNU readline is no longer supported.  Only the built-in readline implementation is.

nealmcb, 2019-09-23:
Thank you, @dpgeorge! And it's good to know about the modal behavior `C-a`, putting you in RAW mode only at the beginning of the line. That probably led to some of my confusion.

I should have also linked to the documentation:
 * [The MicroPython Interactive Interpreter Mode \(aka REPL\) — MicroPython 1\.11 documentation](https://docs.micropython.org/en/latest/reference/repl.html)

But it doesn't seem to list all the keys that MicroPython handles, which are actually often Emacs-like:
  `Ctrl- a b d e f i k n p u`

Except when they aren't (some are in both categories since they are modal):
  `Ctrl- a c d e i u`

I'll also note the Python3 documentation:
* [14\. Interactive Input Editing and History Substitution — Python 3\.7\.4 documentation](https://docs.python.org/3.7/tutorial/interactive.html)

which in turn links to [The GNU Readline Library](https://tiswww.case.edu/php/chet/readline/rltop.html)

I think it would be very helpful if the control keys were all documented at that REPL documentation page, along with their modal aspects.  I dare say I didn't get them all.  A link to the relevant source files here might help in accomplishing that.

dpgeorge, 2019-09-26:
> I think it would be very helpful if the control keys were all documented at that REPL documentation page, along with their modal aspects.

+1