micropython/micropython #5137: micropython can support Apple homekit？

coffice12, 2019-09-23:


dpgeorge, 2019-09-27:
It sounds like this is a feature request.  Please can you provide more detail so progress can be made.

ukd1, 2021-03-11:
@dpgeorge seems likely this should be an external library, not core? but at least my request would be to have some kind of library (core or otherwise) that supported HomeKit protocol. Python library examples are here:

* https://github.com/ikalchev/HAP-python - issue tracking it ikalchev/HAP-python#286
* https://github.com/jlusiardi/homekit_python
* https://github.com/condemil/pyhap

I may port one at some point, but tbh I'm not a Python person really.

MarByteBeep, 2023-12-20:
What's the current status on this one? It would be really cool if for Apple Homekit would be supported on Micropython driven devices, for obvious reasons. I fail to believe that my smart light bulb is actually smarter than for instance a Raspberry Pico W :)

jonnor, 2024-07-27:
Some work has been done in https://github.com/katadelos/micropython-esp32-homekit
In general, this seems to be doable with either a Python or C module - so it does not have to happen in core MicroPython repository. Proposing to move this issue to a discussion.