micropython/micropython #5140: Special treatment of null qstr means "NULL" is not a valid in mpy-cross.

nevercast, 2019-09-25:
Minimum repro:

Create a file with the contents:
```
def expode():
    return "NULL"
```

Add this as a frozen file so that it is compiled with `mpy-cross` and output in to `frozen.c`. 
`frozen.c` now conflicts with `qstr.generated.h` with a duplicate `MP_QSTR_NULL` definition.

Seems to work fine in the REPL, presumably because the string is not QSTRd

```
build-wscale/frozen_mpy.c:764:5: error: redeclaration of enumerator 'MP_QSTR_NULL'
     MP_QSTR_NULL,
     ^
In file included from ../../py/obj.h:31:0,
                 from ../../py/objint.h:30,
                 from build-wscale/frozen_mpy.c:2:
build-wscale/genhdr/qstrdefs.generated.h:3:6: note: previous definition of 'MP_QSTR_NULL' was here
 QDEF(MP_QSTR_NULL, (const byte*)"\x00\x00\x00" "")
      ^
../../py/qstr.h:41:23: note: in definition of macro 'QDEF'
 #define QDEF(id, str) id,
                       ^
../../py/mkrules.mk:47: recipe for target 'build-wscale/build-wscale/frozen_mpy.o' failed
make: *** [build-wscale/build-wscale/frozen_mpy.o] Error 1
```

pmp-p, 2019-09-25:
nice catch

pmp-p, 2019-09-25:
afaik the QSTR parser is run first, then mpy-cross pass is applied but i don't think there's a filter for duplicates removal on the resulting file.

nevercast, 2019-09-25:
`MP_QSTR_NULL` is a special value, it doesn't mean "NULL" it means missing / empty value, I found this out when exploring why my code breaks. So its not really a duplicate result. I'm not sure `__COUNTER__` or `__LINE__` will help in this case as the QSTRs need to be generated across all files. However I'm not intimately familiar with the interning process.

I'll wait for Damien to provide a suggestion, if we want to more intelligently handle the empty enum value, or handle "NULL" as a special case, or just ignore it all together. Though I'd prefer that valid strings didn't break frozen code generation.

pmp-p, 2019-09-25:
indeed i had a look it seems the only special value. ie where "NULL" is not reproductible hash of real target value. looks like a corner case.

pmp-p, 2019-09-25:
for anyone needed a quick temporart fix :
just freeze instead:
`def expose():
    return b"NULL".decode()
`

pmp-p, 2019-09-26:
May i suggest making a tag [breaking changes?] available note the '?' for being nice to people who may *already* use MP_QSTR_NULL in their C/C++ module so they can access information, request more, and overall some time to prepare themselves ?

nevercast, 2019-09-26:
In general MicroPython does not have semantic versioning and does not guarentee a fixed C/C++ API. Breaking changes are to be expected.

commit 7d58a197 (2019-09-26): py: Rename MP_QSTR_NULL to MP_QSTRnull to avoid intern collisions.

Fixes #5140.
closed 2019-09-26 by dpgeorge