micropython/micropython #5145: [STM32WB] USBDONGLE_WB55: I2C

dmazzella, 2019-09-25:
I tried to communicate with several I2C slave devices (ATECC508A and ST25DV04K) but without success.

hw:
```
>>> i2c = machine.I2C (1, freq=400000)
>>> i2c.scan()
[]
>>>
```

sw:
```
>>> i2c = machine.I2C (-1, scl=machine.Pin.cpu.B8, sda=machine.Pin.cpu.B9, freq=400000)
>>> i2c.scan()
[]
>>>
```

can anyone communicate with i2c devices with the stm32wb?

dpgeorge, 2019-09-26:
I didn't test this feature but it's surprising that the software I2C is not working.  Do you have pull-ups on the I2C lines?  Does basic GPIO work?

dmazzella, 2019-09-26:
> I didn't test this feature but it's surprising that the software I2C is not working. Do you have pull-ups on the I2C lines? Does basic GPIO work?

Pull-up did the tricks.

closed 2019-09-26 by dmazzella