micropython/micropython #5147: Use Travis CI cache to backup downloaded dependencies

nevercast, 2019-09-26:
Network problems, SSL keys expiring, domains going down due to a temporary outage can cause tests to fail.

Currently our external web dependencies seem to be:

## ESP32
* wget https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
* git clone https://github.com/espressif/esp-idf.git

## ESP8266
* wget https://github.com/jepler/esp-open-sdk/releases/download/2018-06-10/xtensa-lx106-elf-standalone.tar.gz

I think we can at least download and unpack the xtensa toolchain in to `$HOME/persist` and optionally download them if they are missing. This makes us resistant to `dl.espressif.com` failure.

We should also consider cloning `esp-idf` in to `~/persist` and just fetching if our ESP_IDF hash is missing.

The less connected parts, the better.

dpgeorge, 2019-09-26:
> I think we can at least download and unpack the xtensa toolchain in to `$HOME/persist` and optionally download them if they are missing.

Sounds OK to me, it would speed up builds as well (downside is it'd complicate .travis.yml somewhat).

> This makes us resistant to `dl.espressif.com` failure.

The current problem with that domain is an expired SSL cert, which should be a rare thing...

> We should also consider cloning `esp-idf` in to `~/persist` and just fetching if our ESP_IDF hash is missing.

IMO it's generally better to do things from scratch, to minimise any troubles due to stale things lying around from a previous build.

jonnor, 2023-07-30:
Travis CI is no longer being used. Is this still relevant for the Github Actions CI ? If so, proposing to rename issue to reflect.

mattytrentini, 2023-07-31:
No longer relevant; containers for the development environments are the better solution.

closed 2023-07-31 by mattytrentini