micropython/micropython #5157: Neopixel timing error

klauweg, 2019-09-28:
With the latest ESP32 micropython firmware (esp32-20190928-v1.11-361-g4ba0aff47.bin) the following code fails:

```python
from machine import Pin
from neopixel import NeoPixel
from time import sleep

pin = Pin(22, Pin.OUT)
np = NeoPixel(pin, 60)
np[50] = (5,5,5)
np.write()

while True:
    np.write()
```

While executing the loop we can see randomly flashing some pixels.

I've checked the ws2812 data signal with a scope and discovered that the high pulse of the "0 code" (should be less than 400ns) sometimes extends to 600ns.

esp32-20190529-v1.11.bin shows the same behaviour.

It is running on esp32doit-devkit-v1 hardware.

dpgeorge, 2019-09-30:
Interrupts should be disabled while writing neopixel data, but it could be that something is still interrupting the output (somehow...), or that it's a non-deterministic caching effect.

It's probably a good idea to instead get the RMT peripheral working for driving neopixels.

carstenblt, 2019-10-17:
I get the same behavior with the recent version using the v4 esp-idf.

I see the same behavior using the RMT implementation from the loboris port, which is really, really weird. (it was working before) I don't have a scope to check it unfortunately.

CelebratedPooper, 2019-11-28:
@klauweg Did you find a work around?

xmoiduts, 2019-12-27:
I also encountered this problem when I upgraded from **1.10**(No problem) to **1.12**(Has this problem), the issue exists on both the official release of esp32-idf3 and esp32-idf4 variants and **1.11** also had the same problem. 
Since my LED strip(folded as an LED screen) has 300-LEDs, the interrupt issue in the above post is perhaps the reason.

@CelebratedPooper Did you find a workaround?
@UnexpectedMaker has mentioned successful driving in this [link](https://github.com/micropython/micropython/pull/5184#issuecomment-539801611), Did it work well or also has random subpixel flashing issue?

UnexpectedMaker, 2019-12-27:
There is no solution for ESP32 for chaining single data line variants of addressable RGB LEDs longer than 14 (if RGB) and 9 (if GRBW) without using RMT. It's not specific to MicroPython - It's IDF and Arduino too. 

1.12 now ships with RMT support, and I have an RMT based driver that I am working on that I will be releasing *soon* that I've driven 175x GRBW pixels with no issues.

I have a [previous live stream](https://youtu.be/h4df0gyMMAs) about it where I get it all working.

xmoiduts, 2019-12-27:
What's the reason for the 14(GRB) or 9(GRBW) LED limit without RMT? Is there some link for the information? 

Also, the appearance of my LED panel has been added in previous comment.

UnexpectedMaker, 2019-12-27:
It's about the most data it can bit bang out between that interrupt firing messing with the data.
Less for GRGW LEDs as they have the extra white data being sent, so 4x the colour info instead of 3.

The 14 and 9 are not exact numbers, but are pretty close to my experiences of issues. Sometimes it's 11 and 7.

There is plenty of info about all of this if you just google ESP32 neopixel corruption or similar.

wangshujun-tj, 2020-01-08:
I have tried to drive more than 100 ws2812 RGB lights with the SPI bus of esp32, but the process of building data is not very comfortable, because using a byte of SPI to transmit one bit takes up a lot of RAM space, and when the data transmission volume of SPI bus is large  DMA will be used automatically, so the timing is safe

xmoiduts, 2020-01-09:
> I have tried to drive more than 100 ws2812 RGB lights with the SPI bus of esp32, but the process of building data is not very comfortable, because using a byte of SPI to transmit one bit takes up a [...]

Have you also encountered this flickering issue with pure CPU bit-banging after a certain version of micropython? Since the per-millisecond interruption was suspected to be the reason, I'm confused that why it didn't happen in elder firmware versions. Didn't they use such interrupts or the interrupts didn't happen on the micropython core?

I have read the documentation of current micropython RMT API(very briefly) and realized that there will be the same byte-for-bit memory issue, which, on a 300-LED panel, consumes > 10KB out of a precious 100-ish KB total memory.

thijstriemstra, 2020-01-19:
>  I have an RMT based driver that I am working on that I will be releasing soon 

any news on this @UnexpectedMaker, is there something we can help test?

UnexpectedMaker, 2020-01-19:
Hey @thijstriemstra 
@nevercast created this, so pressure is off me to "finish" my implementation :)
https://gist.github.com/nevercast/9c48505cc6c5687af59bcb4a22062795

Fjordx, 2020-07-04:
Although the grounds were connected before( pc and usb power supply) I had hardcore timing errors till i decided to feed the esp and the neopixels from the same powersupply. Then everything worked flawlessly. dont forget the resistor on the dataline, was crucial for me as well.

Wei1234c, 2021-05-10:
I had the randomly flashing problem.

Then I found  [The WS2812B RGB LED has been revised](https://blog.adafruit.com/2017/05/03/psa-the-ws2812b-rgb-led-has-been-revised-will-require-code-tweak/), so I simply hold the data pin low after np.write(). 

The problem is gone.

jimmo, 2021-11-11:
Please see https://github.com/micropython/micropython/pull/7985 for an RMT-based neopixel driver for ESP32.

dpgeorge, 2021-12-22:
The esp32 neopixel driver now uses RMT, see 599b61c08687ca077e3b0e115d5b76affcc673ca.  That should fix the timing issues.

closed 2021-12-22 by dpgeorge