micropython/micropython #5159: esp32: ULP binary loading support was broken after 1.10 release

ThomasWaldmann, 2019-09-29:
See there for details and code to reproduce the issue:

https://github.com/ThomasWaldmann/py-esp32-ulp/issues/42

Works: micropython 1.10

Broken: micropython 1.11 (and also latest daily build downloaded today)

dpgeorge, 2019-09-30:
Thanks for the report.  There are no tests for ULP (or much of the esp32's specific functionality) so that's why it regressed.

ThomasWaldmann, 2019-09-30:
Are the tests run on real esp32 hardware?

bruno963852, 2019-10-02:
I'd really like to use ULP and the new BLE functionality that was just merged :-(

mattytrentini, 2019-11-04:
There's nothing obviously of note that's changed in the MicroPython ULP code since v1.11; I wonder if it's due to an update in the ESP-IDF?

dpgeorge, 2019-12-05:
Fixed by 9ca8a503ed3d7b31fc1cfe0e2e51925d3c0c3396

closed 2019-12-05 by dpgeorge