micropython/micropython #5162: [nRF port] Unable to activate MICROPY_PY_SYS_STDFILES in mpconfigport.h

sebi5361, 2019-09-30:
I was unable to turn on some of the features in `mpconfigport.h`, such as `MICROPY_PY_SYS_STDFILES`, `MICROPY_PY_MATH_SPECIAL_FUNCTIONS`, `MICROPY_PY_IO_FILEIO` and `MICROPY_PY_MACHINE_SPI`, as `make` would fail with those features on (nRF port).

Below are the error messages:
- Error when `#define MICROPY_PY_SYS_STDFILES (1)`:
```
LINK build-pca10040-s132/firmware.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccCsRh6Z.ltrans2.ltrans.o: in function `mp_sys_print_exception.lto_priv.992':
/mnt/c/Users/krut/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modsys.c:110: undefined reference to `mp_sys_stdout_obj'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccCsRh6Z.ltrans2.ltrans.o:(.rodata+0xb4): undefined reference to `mp_sys_stdout_obj'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccCsRh6Z.ltrans2.ltrans.o:(.rodata+0x1c8): undefined reference to `mp_sys_stdout_obj'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccCsRh6Z.ltrans2.ltrans.o:(.rodata+0x246c): undefined reference to `mp_sys_stdin_obj'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccCsRh6Z.ltrans2.ltrans.o:(.rodata+0x2474): undefined reference to `mp_sys_stdout_obj'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccCsRh6Z.ltrans2.ltrans.o:(.rodata+0x247c): undefined reference to `mp_sys_stderr_obj'
collect2: error: ld returned 1 exit status
make: *** [Makefile:344: build-pca10040-s132/firmware.elf] Error 1
```
- Error when `#define MICROPY_PY_MATH_SPECIAL_FUNCTIONS (1)`
```
CC build-pca10040-s132/pins_gen.c
LINK build-pca10040-s132/firmware.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_lgamma.lto_priv.1007':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:170: undefined reference to `lgammaf'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_gamma.lto_priv.1006':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:168: undefined reference to `tgammaf'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_erfc.lto_priv.1005':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:166: undefined reference to `erfcf'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_erf.lto_priv.1004':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:164: undefined reference to `erff'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_atanh.lto_priv.982':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:120: undefined reference to `atanhf'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_asinh.lto_priv.981':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:118: undefined reference to `asinhf'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_acosh.lto_priv.980':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:116: undefined reference to `acoshf'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/cchaRFoM.ltrans2.ltrans.o: in function `mp_math_log2.lto_priv.975':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:106: undefined reference to `log2f'
collect2: error: ld returned 1 exit status
make: *** [Makefile:344: build-pca10040-s132/firmware.elf] Error 1
```
- Error when `#define MICROPY_PY_IO_FILEIO (1)`
```
LINK build-pca10040-s132/firmware.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccKalfRU.ltrans2.ltrans.o:(.rodata+0x25fc): undefined reference to `fatfs_type_fileio'
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccKalfRU.ltrans2.ltrans.o:(.rodata+0x2604): undefined reference to `fatfs_type_textio'
collect2: error: ld returned 1 exit status
make: *** [Makefile:344: build-pca10040-s132/firmware.elf] Error 1
```
- Error when `#define MICROPY_PY_MACHINE_SPI (1)`:
```
CC ../../extmod/machine_spi.c
../../extmod/machine_spi.c: In function 'mp_machine_soft_spi_print':
../../extmod/machine_spi.c:175:67: error: expected ')' before 'MP_HAL_PIN_FMT'
     mp_printf(print, "SoftSPI(baudrate=%u, polarity=%u, phase=%u,"
                                                                   ^
                                                                   )
         " sck=" MP_HAL_PIN_FMT ", mosi=" MP_HAL_PIN_FMT ", miso=" MP_HAL_PIN_FMT ")",
                 ~~~~~~~~~~~~~~
../../extmod/machine_spi.c:174:32: error: unused variable 'self' [-Werror=unused-variable]
     mp_machine_soft_spi_obj_t *self = MP_OBJ_TO_PTR(self_in);
                                ^~~~
At top level:
../../extmod/machine_spi.c:146:17: error: 'baudrate_from_delay_half' defined but not used [-Werror=unused-function]
 STATIC uint32_t baudrate_from_delay_half(uint32_t delay_half) {
                 ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:47: build-pca10040-s132/extmod/machine_spi.o] Error 1
```

dpgeorge, 2019-09-30:
>  MICROPY_PY_SYS_STDFILES

For this it'll also need the file `lib/utils/sys_stdio_mphal.c` added to the build via the Makefile.

> MICROPY_PY_MATH_SPECIAL_FUNCTIONS

For this there will be additional files from `lib/libm/` needed.

sebi5361, 2019-09-30:
> > MICROPY_PY_SYS_STDFILES

> For this it'll also need the file `lib/utils/sys_stdio_mphal.c` added to the build via the Makefile.

In the `Makefile`, I have added `utils/sys_stdio_mphal.c` to `SRC_LIB`:
```
SRC_LIB += $(addprefix lib/,\
	libc/string0.c \
	mp-readline/readline.c \
	utils/pyexec.c \
	utils/interrupt_char.c \
	timeutils/timeutils.c \
	utils/sys_stdio_mphal.c \
	)
```
and `$(SRC_LIB)` to `SRC_QSTR`:
```
# List of sources for qstr extraction
SRC_QSTR += $(SRC_C) $(DRIVERS_SRC_C) $(SRC_LIB) $(SRC_BOARD_MODULES)
```
But still I am getting an error:
```
LINK build-dsd6-s132/firmware.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccUFd8Pz.ltrans0.ltrans.o: in function `stdio_ioctl.lto_priv.390':
/mnt/c/Users/krut/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../lib/utils/sys_stdio_mphal.c:91: undefined reference to `mp_hal_stdio_poll'
collect2: error: ld returned 1 exit status
make: *** [Makefile:345: build-dsd6-s132/firmware.elf] Error 1
```
Can you tell me how to modify the `Makefile` exactly?

sebi5361, 2019-09-30:
> > MICROPY_PY_MATH_SPECIAL_FUNCTIONS
> 
> For this there will be additional files from `lib/libm/` needed.

I have added those files to `SRC_LIB` in the `Makefile`:
```
SRC_LIB += $(addprefix lib/,\
        ...
	libm/erf_lgamma.c \
	libm/wf_lgamma.c \
	libm/wf_tgamma.c \
	libm/atanhf.c \
	libm/log1pf.c \
	libm/asinhf.c \
	libm/acoshf.c \
	libm/sf_erf.c \
	)
```
But I still get this error:
```
LINK build-pca10040-s132/firmware.elf
/usr/local/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: /tmp/ccQBuPBs.ltrans2.ltrans.o: in function `mp_math_log2.lto_priv.968':
/mnt/c/Users/sebi/Documents/TINKERING/MicroPython/FORKS/micropython/ports/nrf/../../py/modmath.c:106: undefined reference to `log2f'
collect2: error: ld returned 1 exit status
make: *** [Makefile:353: build-dsd6-s132/firmware.elf] Error 1
```
What additional file should I add to get rid of the `log2f` undefined reference error?

dpgeorge, 2019-10-08:
I fexd the SYS_STDFILES build in 2863dcdf4f9f1fc421c7b745195259fbf46250a2

dpgeorge, 2021-06-08:
The math special functions issue was solved in 8c02b94946c437ed841fe294442b1f0d2401c678