micropython/micropython #5163: [nRF port] Unable to Activate the FAT File System

sebi5361, 2019-09-30:
I noticed that FAT file system couldn't be enabled on a nRF board as `open("code.py", "wb")` would fail making it impossible to create files onto the board.

Indeed, after building and flashing the firmware with the FAT file system enabled
(`#define MICROPY_VFS (1)` in `mpconfigport.h` and `make SD=s132 MICROPY_FATFS=1 sd`) the following error occurs:
```
>>> open("code.py", "wb")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
```

glennrub, 2019-09-30:
Hi @sebi5361 ,

I can reproduce this. I assume you have also enabled `MICROPY_VFS=1`. 
I'm pretty sure this is related to `microbitfs` not being compatible with VFS. There has been work by @aykevl on this.

@aykevl, would you like to pick this up? Or could i continue where you left?

sebi5361, 2019-09-30:
Yes indeed, the upper content of `mpconfigport.h` is:
```
#ifndef MICROPY_VFS
#define MICROPY_VFS                 (1) // Append MICROPY_FATFS=1 to `make` when (1)
#endif
#define MICROPY_VFS_FAT             (MICROPY_VFS)
```

mzdaniel, 2019-12-06:
Seems that this issue is also preventing little fs to be used on nrf. When adding vfs_lfs (and verifying it properly works on ramdisk [https://github.com/micropython/micropython/blob/master/docs/reference/filesystem.rst#custom-block-devices]), flash access stops working properly:

pyboard -f cp README.md :
cp README.md :README.md
Traceback (most recent call last):
  File "<stdin>", in <module>
OSError: 1

As far as I can tell, enabling VFS breaks flash on nrf