micropython/micropython #5164: PSA: Incompatible .mpy version

nevercast, 2019-09-30:
There was a recent version bump in the MPY files as the bytecode header has changed. https://github.com/micropython/micropython/commit/5716c5cf65e9b2cb46c2906f40302401bdd27517.  When running make after changing some frozen python files I was getting the following error:

```
Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 815, in <module>
    main()
  File "../../tools/mpy-tool.py", line 803, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 803, in <listcomp>
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 657, in read_mpy
    raise Exception('incompatible .mpy version')
Exception: incompatible .mpy version
```

This issue is a PSA for how I fixed this, simply clean your build, rebuild `mpy-cross` and then rebuild your port. For example:

`~/micropython$ make -C ports/esp32` failed with the above error

Rebuild `mpy-cross`

`~/micropython$ make -C mpy-cross`

Clean your port 

`~/micropython$ make -C ports/esp32 clean`

Rebuild your port

`~/micropython$ make -C ports/esp32`

Now your mpy files will be correctly built with the latest `mpy-cross` and they will not fail a version check later in the build process. I will now close this issue so that it sits as a reference.

closed 2019-09-30 by nevercast

peterhinch, 2019-10-01:
This is something of a FAQ: see [this script](https://github.com/peterhinch/micropython-samples/blob/master/fastbuild/buildnew). Perhaps the docs need improving.

mcauser, 2019-10-10:
Thanks Josh. I also needed to run a `make -C mpy-cross clean` to get rid of the old version.