micropython/micropython #5169: ESP32:  add support for 160MHz rated CPUs

fgiava, 2019-10-01:
Some recent ESP32 chips are rated only to 160MHz: ESP32-D2WD or ESP32-S0WD.

Flashing this kind of ESP32 board with the current mPython version 1.11 we have a continuous reset loop with primary bootloader activation.

The same problem has been reported in the Arduino environment “Espressif ESP32 board support” and solved in the last release 1.03.
Here some relevant link:

https://forum.micropython.org/viewtopic.php?f=18&t=6980&p=39922&hilit=240#p39922

https://esp32.com/viewtopic.php?f=12&t=11439&start=10

dpgeorge, 2019-10-08:
The way to progress with this would be to separate out the `CONFIG_ESP32_DEFAULT_CPU_FREQ_240` option from `sdkconfig.base` and put it in a separate file.  Then the boards can easily configure their frequency.  Probably a good idea to default to 160MHz for the `GENERIC_D2WD` board.

nevercast, 2019-10-09:
The usual default is boot to 80MHz is this not the case for MicroPython?

dpgeorge, 2019-10-10:
> The usual default is boot to 80MHz is this not the case for MicroPython?

No, it's 240MHz.  In sdkconfig.base there is the following line:
```
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
```

dpgeorge, 2019-10-18:
Implemented in 59c1b7d4b8fb1d441f9322976582c8709f01dfbe

closed 2019-10-18 by dpgeorge