micropython/micropython #5173: false GPIO interrupt triggered on ads.read()

nedoskiv, 2019-10-02:
Hello,

first I want to say i'm using v1.10 on ESP32 (perhaps someone already found&reported this and it is fixed on latest).
I got a small project to read 2 analog sensor, ds18b20 and a pulse device. Decided to use gpio interrupt to count pulse device, and then found that false irq request are triggered. Isolating part of code below, found that false call on gpio callback function is triggered on adc.read().

> i={}
> z={}
> import machine
> i["s1"] = machine.ADC(machine.Pin(34))
> i["s1"].atten(machine.ADC.ATTN_11DB)
> i["s2"] = machine.ADC(machine.Pin(35))
> i["s2"].atten(machine.ADC.ATTN_11DB)
> 
> 
> i["inc"]=False	# increment flag, if set up, mean main loop not updated first result yet
> z["c"]=0	# vodomer counter
> def pi(p):
> 	print ("Interrupt!")
> 	global z,i
> 	if i["inc"]==False:
> 		i["inc"]=True
> 		z["c"]+=1
> i["v"] = machine.Pin(26, machine.Pin.IN, machine.Pin.PULL_UP)		# vodomer pulse gpio
> i["v"].irq(trigger=machine.Pin.IRQ_FALLING, handler=pi)
> 
> 
> import onewire, ds18x20, time
> 
> i["ow"] = machine.Pin(4)
> i["ds_sensor"] = ds18x20.DS18X20(onewire.OneWire(i["ow"]))
> 
> i["roms"] = i["ds_sensor"].scan()
> z["s1"]=0
> z["s2"]=0
> z["ds"]="unset"
> print('Found DS devices: ', i["roms"])
> i["s2"].read()
> i["s2"].read()
> i["s2"].read()
> i["s2"].read()
> i["s2"].read()

nedoskiv, 2019-10-02:
P.S. Change pin where I attach interrupt to 15 and it work fine. (as long as I remember there was 2 adc "channels" on esp32, one is usable for user code, second was not usable becouse of the wireless ........

dpgeorge, 2019-10-08:
This sounds like a hardware limitation, or problem at the ESP IDF level.  So I'll close the issue.  Feel free to reopen/comment if it can be shown that it's an issue with MicroPython.

closed 2019-10-08 by dpgeorge