micropython/micropython #5174: ble support and encrypted characteristics

jonsmirl, 2019-10-02:
Are there plans for supporting encrypted characteristics in the new BLE API?  In my use case I set a password into the BLE device and I don't want to send it in the clear. So my characteristic needs to be marked encrypted.

Not sure if this involves much code at the python level, it should be more along the lines of pushing the right flags down into the various implementations. The stacks will take care of the encrypt/decrypt.

jimmo, 2019-10-06:
@jonsmirl Yes this is definitely on my radar, like you say it should hopefully be fairly straightforward. I'm going to raise a "BLE roadmap" tracking issue. I also plan to implement pairing & bonding, which obviously relate to this feature too.

dmazzella, 2019-11-18:
> @jonsmirl Yes this is definitely on my radar, like you say it should hopefully be fairly straightforward. I'm going to raise a "BLE roadmap" tracking issue. I also plan to implement pairing & bonding, which obviously relate to this feature too.

@jimmo @dpgeorge pairing & bonding is in progress? if not, are there time plans?