micropython/micropython #5176: The port for pic16bit is broken.

jacklinquan, 2019-10-03:
I've confirmed my findings with the micropython port for pic16 from [url]https://github.com/micropython/micropython/tree/master/ports/pic16bit[/url]

I built this port with PIC24FJ1024GB606 and the same Makefile(the path of XC16 changed).
You can find it here: [url]https://github.com/jacklinquan/micropython/tree/master/ports/pic24fj1024gb606[/url]

Here's some brief test result:
When NDEBUG is defined in the Makefile:
##################
>>> 1+1
2
>>> a=1
>>> a
1
>>> b=2
>>> c=a+b
>>> c
Traceback (most recent call last):
  File "<stdin>", in <module>
NameError: name not defined
>>> 256*256
Traceback (most recent call last):
  File "<stdin>", in <module>
TypeError: unsupported type for operator
>>> 25*25
625
##################

When NDEBUG is NOT defined in the Makefile:
##################
>>> 1+1
2
>>> a=1
assertion "exc_sp == exc_stack - 1" failed: file "../../py/vm.c", line 1143, function: mp_execute_bytecode
MicroPython v1.11 on 2019-10-03; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> 25*25
625
>>> 256*256
assertion "emit->stack_size == 0" failed: file "../../py/emitbc.c", line 426, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-03; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> dir
<function>
>>> dir()
assertion "emit->stack_size == 0" failed: file "../../py/emitbc.c", line 426, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-03; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> int
<class 'int'>
>>> int('1')
assertion "emit->stack_size == 0" failed: file "../../py/emitbc.c", line 426, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-03; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> s='hello'
assertion "exc_sp == exc_stack - 1" failed: file "../../py/vm.c", line 1143, function: mp_execute_bytecode
MicroPython v1.11 on 2019-10-03; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> l=[1,2,3]
assertion "emit->stack_size == 0" failed: file "../../py/emitbc.c", line 426, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-03; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
##################

For the moment I just have PIC24FJ1024GB606 at hand. Could anyone help confirm my findings with your boards? 
It looks like the port for pic16bit is broken.

Thanks!

jacklinquan, 2019-10-04:
Some update on this issue:
With the latest update of master branch, the 'assertion "exc_sp == exc_stack - 1" failed' problem is gone.

Here's the updated test result:(without -DNDEBUG compiler flag)
'''
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606
>>> 1+1
2
>>> a=1
>>> a
1
>>> b=2
>>> b
2
>>> c=a+b
assertion "emit->stack_size == 0" failed: file "py/emitbc.c", line 400, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> 25*25
625
>>> 256*256
assertion "emit->stack_size == 0" failed: file "py/emitbc.c", line 400, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> int
<class 'int'>
>>> int('1')
assertion "emit->stack_size == 0" failed: file "py/emitbc.c", line 400, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> dir
<function>
>>> dir()
assertion "emit->stack_size == 0" failed: file "py/emitbc.c", line 400, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> s='hello'
>>> s
'hello'
>>> s[0]
assertion "emit->stack_size == 0" failed: file "py/emitbc.c", line 400, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> print
<function>
>>> print('hello')
assertion "emit->stack_size == 0" failed: file "py/emitbc.c", line 400, function: mp_emit_bc_end_pass
MicroPython v1.11 on 2019-10-04; Experiment with PIC24FJ1024GB606 (Note: It rebooted here.)
'''

dpgeorge, 2019-10-04:
Commit 6d480f50accb475a67ce53a6cbbaccea603e59ed was the most recent one which was verified to run on hardware, on a dsPIC33FJ256GP506

So please try that commit.  If it still has problems then most likely there's an issue with it working on the PIC24F.

jacklinquan, 2019-10-04:
Thank you a lot @dpgeorge !
I can try to get a dsPIC33FJ256GP506 dev board to confirm it. But it could take quite some time.
Before I get this board, could you shed some light on how I could work around to fix this issue for PIC24F?
It would be great if there is any detailed doc on how micropython stack and vm work. With the docs I can try to figure it out by myself.

dpgeorge, 2019-10-04:
This PIC 16-bit architecture is difficult to use, because it has ~1Mb flash and ~32k RAM, but only a 16-bit address space.  So there are compiler/linker tricks to address all the memory space.  It may be that those tricks are failing on the PIC24F.

In MicroPython ROM and RAM objects must be accessible in the same address space, because a lot of constant objects live in ROM/flash.  This compounds the problems running it on a 16-bit PIC (but it's still possible, at least on the dsPIC33F).

It's hard to debug this stuff.  You'll need to look in the linker map and understand exactly the memory layout and how entities are addressed, and trace through the execution.

(IMO it's much easier to just use a 32-bit architecture.)

jacklinquan, 2019-10-04:
Hi @dpgeorge ,

Thank you for the in-depth explanation. Now I have much better understanding on where this issue comes from. Maybe it's time to move on to a 32-bit architecture.

Anyway here's my test result with PIC24FJ1024GB606 on commit 6d480f5 :
```
MicroPython v1.9.4 on 2019-10-04; EXPERIMENT with PIC24FJ1024GB606
>>> 1+1
2
>>> 25*25
625
>>> 256*256
assertion "emit->stack_size == 0" failed: file "../../py/emitbc.c", line 413, function: mp_emit_bc_end_pass
MicroPython v1.9.4 on 2019-10-04; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> a=1
assertion "exc_sp == exc_stack - 1" failed: file "../../py/vm.c", line 1093, function: mp_execute_bytecode
MicroPython v1.9.4 on 2019-10-04; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
>>> s='hello'
assertion "exc_sp == exc_stack - 1" failed: file "../../py/vm.c", line 1093, function: mp_execute_bytecode
MicroPython v1.9.4 on 2019-10-04; EXPERIMENT with PIC24FJ1024GB606 (Note: It rebooted here.)
```

dpgeorge, 2019-10-04:
> Anyway here's my test result with PIC24FJ1024GB606 on commit 6d480f50accb475a67ce53a6cbbaccea603e59ed

So it seems the PIC24F itself is the problem.

Note the dsPIC33FJ256GP506 has only 16k RAM so there's 48k left to map part of flash, and the linker will place rodata there so it can be addressed in 16-bits.  The PIC24F has 32k RAM so it could be that 32k left over is not enough for all the rodata to be addressed in 16-bits.  It might be possible to reduce the available RAM to 16k (using a different linker script, or select a part that is similar but has less RAM) to get it working.

jacklinquan, 2019-10-04:
Thank you @dpgeorge !
I will try it for sure once I get a PIC24F micro that has less RAM, and then update my test result here.